gopro/OpenGoPro #435: Bug Description

adilerson, 2023-11-28:
**Component**
What is the bug in?
- Specs (HTTP)

**Describe the bug**
list files and keep alive stop working

**To Reproduce**
Steps to reproduce the behavior:
1. Using Linux, I connect to the GoPro’s WiFi
2. I send a signal, start recording with http shutter start and end with stop, everything ok
3. I can list the files and download 2x, after this some functions stop working
4. So far I noticed that keep_alive and media/list stop working

**Expected behavior**
trying to make a quick connection via bluetoothctl:
[bluetooth]# connect CC:9C:B1:39:AA:DA
Attempting to connect to CC:9C:B1:39:AA:DA
[CHG] Device CC:9C:B1:39:AA:DA Connected: yes
Connection successful
[CHG] Device CC:9C:B1:39:AA:DA ServicesResolved: yes
[GoPro 9709]# pair CC:9C:B1:39:AA:DA
Attempting to pair with CC:9C:B1:39:AA:DA
Failed to pair: org.bluez.Error.ConnectionAttemptFailed

**Screenshots**

**Hardware**
- Camera: Hero12.
- Linux Ubuntu server

**Additional context**

if (isset($_GET['downloadLast'])) {
     $base = "http://10.5.5.9:8080";
     $list =
     $destinationdirectory = '/var/www/html/360Linux/360v2/celular/videos/tmp/';
     $currentevent = currentevent();
     $list = "$base/gopro/media/list";
     $url = $lista;
     **$json = file_get_contents($url);** // espta stops working
     $data = json_decode($json, true);
    
    

     // Check if decoding was successful
     if ($data !== null) {
         // Initialize variables for the most recent video
         $videoMostRecent = null;
         $timestampMostRecent = 0;
    
         // Iterate over all folders and videos to find the latest
         foreach ($data['media'] as $folder) {
             foreach ($folder['fs'] as $video) {
                 $timestamp = intval($video['cre']);
                 if ($timestamp > $timestampMostRecent) {
                     $timestampMostRecent = $timestamp;
                     $filefolder = $folder['d'];
                     $videoMostRecent = $video;
                 }
             }
         }

         if ($videoMostRecent !== null) {

             $fileName = $videoMostRecent['n'];
             $filefolder = $data['media'][0]['d'];
             $urlCompleta = $base . '/videos/DCIM/' . $filefolder . '/' . $fileName;
            
             $filenameNew = pathinfo($filename);
             $nomeNoExtensao = $filenomeNovo['filename'];
             $completeFilepath = $destinationdirectory. '/' . $nomeNoExtensao.'.mp4';
            
             if (file_exists($FullFilePath)) {
                 echo "File found at: $FullFilePath<br>";
                
             }else{
                 $txt = eventCurrentTxt();
                 if (file_put_contents($ DirectorioDestino . '/' . $nomeSemExtensao . '.txt', $txt) === false) {
                     echo "Error creating text file.<br>";
                 } else {
                     echo $directoryDestino . '/' . $nomeSemExtensao . '.txt<br>';
                     echo "Text file created successfully.<br>";
                 }
                 echo $directoryDestino . '/' . $nomeNoExtensao.'.txt<br>';
            
                 $file_info = pathinfo(basename($url));
                 $newfile = $nomeSemExtensao.'.mp4';
                 $call = file_get_contents($turboOn);
        
                 // Download the file to the destination directory
                 if (copy($Fullurl, $FullFilepath)) {
                     $off = file_get_contents($turboOff);
                     $configQuality = '/var/www/html/360Linux/360v2/config.json';
                     $config_qualidade = json_decode(file_get_contents($configQualidade), true);
                     $qualidade = $config_qualidade['qualidade'] ?? 'faster';
                     ffmpeg($nomeNoExtensao,$preset,$crf,true);
                 } else {
                     echo "Error downloading file.";
                 }
             }
                
         } else {
             echo "No videos found.";
         }
     } else {
         echo "Error decoding JSON.";
     }
}

adilerson, 2023-11-28:
Linux Ubuntu server 22.04

tcamise-gpsw, 2023-12-07:
Can you provide a log or complete script so that I can reproduce this?

closed 2024-01-18 by github-actions[bot]