gopro/OpenGoPro #437: Http response "Unexpected status line" on GoPro 12 Black

Marvel-Android, 2023-12-01:
**Component**
What is the bug in?
GoPro 12 Black(Version 01.30) returns unexpected status line like "gpHttpServer: [04322435][INFO ] conn 0: /gopro/media/list @civetweb.c:17872" in some/all apis.

**Describe the bug**
When use Okhttp, throws this exception:
java.net.ProtocolException: Unexpected status line: gpHttpServer: [04327868][INFO ] conn 0:
/videos/DCIM/100GOPRO/GX012560.MP4 @civetweb.c:17872

When use socket directly, return:
gpHttpServer: [04325594][INFO ] conn 0: /gopro/media/list @civetweb.c:17872
HTTP/1.1 200 OK
Content-Type: application/json
Content-Length: 1255
                                                            
 {"id":"135768295608...

Will this issue be fixed?

tcamise-gpsw, 2023-12-07:
I have not been able to replicate this.
Can you provide either a script to replicate this? 
Or if not, at least the curl commands I can use to replicate it?

Marvel-Android, 2023-12-08:
> I have not been able to replicate this. Can you provide either a script to replicate this? Or if not, at least the curl commands I can use to replicate it?

It happens when I request media list via wired(http://172.28.126.51:8080/gopro/media/list) or wireless(http://10.5.5.9:8080/gopro/media/list) connection.

tcamise-gpsw, 2023-12-08:
Hi. We have reproduced this internally and should have a fix for the next release.

tcamise-gpsw, 2023-12-14:
Unfortunately this didn't make the cutoff for the next release so is delayed until the subsequent release

pavlosharhan2, 2024-02-11:
Hi! is it solved?
Is there any workaround to download the media from Hero12 like older firmware, or Labs? Can't get it working too!

pavlosharhan2, 2024-02-11:
Still getting this debug message on newest firmware for Hero 12 from Feb 8.
Please could we get it fixed, it really confuses all the Android/Python network clients we use...

pavlosharhan2, 2024-02-12:
Here is a working version for python to download a file:

```
import socket

def download_file(host, path, file_name):
    request = f"GET {path} HTTP/1.1\r\nHost: {host}\r\n\r\n"
    with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
        s.connect((host, 80))
        s.send(request.encode())

        with open(file_name, "wb") as file:
            headers_received = False
            response_buffer = b""
            partial_line = b""  # Buffer for incomplete debug line
            removed_lines = []

            while True:
                chunk = s.recv(4096)
                if not chunk:
                    break

                # Check and append partial line to chunk if it exists
                if partial_line:
                    chunk = partial_line + chunk
                    partial_line = b""

                response_buffer += chunk

                # Process and remove all 'gpHttpServer:' lines
                while b"gpHttpServer:" in response_buffer:
                    start = response_buffer.find(b"gpHttpServer:")
                    end = response_buffer.find(b"\n", start) + 1

                    if end > 0:
                        removed_lines.append(response_buffer[start:end].decode())
                        response_buffer = response_buffer[:start] + response_buffer[end:]
                    else:
                        # Store incomplete line and wait for more data
                        partial_line = response_buffer[start:]
                        response_buffer = response_buffer[:start]
                        break

                # Process headers
                if not headers_received:
                    headers_end = response_buffer.find(b"\r\n\r\n") + 4
                    if headers_end > 4:
                        file.write(response_buffer[headers_end:])
                        headers_received = True
                        response_buffer = b""
                elif headers_received:
                    file.write(response_buffer)
                    response_buffer = b""

    for line in removed_lines:
        print("Removed line:", line)

host = "10.5.5.9"
path = "/videos/DCIM/100GOPRO/GX014223.MP4"
file_name = "GX014223.MP4"

try:
    download_file(host, path, file_name)
    print(f"File downloaded successfully as {file_name}")
except Exception as e:
    print(f"An error occurred: {e}")

```

Make a request:

```
import socket

def send_raw_http_request(host, path):
    request = f"GET {path} HTTP/1.1\r\nHost: {host}\r\n\r\n"
    with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
        s.connect((host, 80))
        s.send(request.encode())

        # Read the response in chunks
        response = b''
        while True:
            chunk = s.recv(4096)
            if not chunk:
                break
            response += chunk
    return response

host = "10.5.5.9"
path = "/gopro/media/list"

try:
    response = send_raw_http_request(host, path)
    print("Response received successfully.")
    response_text = response.decode()
    print("Raw response:")
    print(response_text)

    # Check if the response contains JSON
    if "{" in response_text and "}" in response_text:
        json_start = response_text.find("{")
        json_end = response_text.rfind("}") + 1
        json_response = response_text[json_start:json_end]
        print("JSON response:")
        print(json_response)
    else:
        print("No JSON found in response.")

except Exception as e:
    print(f"An error occurred: {e}")
```

Kotlin GET request:

```
fun makeRequest(urlString: String, completion: (Boolean, String?) -> Unit) {
            CoroutineScope(Dispatchers.IO).launch {
                try {
                    withTimeout(10000L) { // 10 seconds timeout
                        val url = URL(urlString)
                        val host = url.host
                        val port = if (url.port != -1) url.port else 80
                        val path = url.path

                        Socket(host, port).use { socket ->
                            socket.soTimeout = 3000 // 3 seconds timeout

                            val request = "GET $path HTTP/1.1\r\n" +
                                    "Host: $host\r\n" +
                                    "Connection: close\r\n\r\n"

                            PrintWriter(socket.getOutputStream(), true).apply {
                                print(request)
                                flush()
                            }

                            BufferedReader(InputStreamReader(socket.getInputStream())).use { reader ->
                                // Skip HTTP headers and read only the JSON response
                                val jsonResponse = reader.lineSequence()
                                    .dropWhile { it.isNotBlank() } // Skip headers
                                    .drop(1) // Skip the empty line after headers
                                    .joinToString("\n")

                                withContext(Dispatchers.Main) {
                                    completion(true, jsonResponse)
                                }
                            }
                        }
                    }
                } catch (e: Exception) {
                    withContext(Dispatchers.Main) {
                        completion(false, null)
                    }
                }
            }
        }

```

pavlosharhan2, 2024-02-12:
UPD: found out that the problem disappears after a Factory reset!
But I've already seen this on multiple cameras, so I'm pretty sure I'll see the issue again

chaneylc, 2024-03-21:
Is there a plan to fix this? I'm running into this randomly only on a GoPro 12 Black. Factory reset did not fix the issue.

JosiSiah, 2024-04-10:
I am also experiencing this issue.

GoPro 12 Black
Python 3.10.14
open-gopro 0.15.1

pavlosharhan2, 2024-04-10:
> I am also experiencing this issue trying it with a Wired Connection.
> 
> GoPro 12 Black Python 3.10.14 open-gopro 0.15.1

@JosiSiah  What is the  firmware version of the GoPro12 ?

JosiSiah, 2024-04-10:
@pavlosharhan2 GoPro version 02.20, updated original comment.

tcamise-gpsw, 2024-04-11:
This is supposed to be fixed on Hero 12 v2.20 and I am unable to replicate it. 
@JosiSiah what endpoint are you using to get the error? Also can you double check your FW version on the camera UI via Preferences ->About -> Camera Info

tcamise-gpsw, 2024-04-11:
Actually please post the full list of calls you are making.  This big is triggered via previous calls to the one that is actually returning the bad response

closed 2024-05-24 by github-actions[bot]