gopro/OpenGoPro #442: docs: wrong name or description on WLAN SSID

alestiago, 2023-12-06:
Status ID value 29 has name "WLAN SSID" but the description details it as "provisioned WIFI **AP** SSID".

On the other hand Status ID value 30 has name "Ap ssid" but the description details it as "Camera’s WIFI SSID".

[image]

---

Source: https://gopro.github.io/OpenGoPro/http_2_0#status-ids

tcamise-gpsw, 2023-12-11:
Actually these are the correct descriptions. To summarize in a different way here:
29 == The SSID of the AP the camera is connected to
30 == The camera’s SSID while in AP Mode.

tcamise-gpsw, 2023-12-11:
We ended up clarifying the status descriptions [here](https://github.com/gopro/OpenGoPro/commit/3ba26fb360ab5cda691b8f05e5013db61a502af1)

closed 2023-12-11 by tcamise-gpsw