gopro/OpenGoPro #449: Arduino BLE Remote

Wumfg, 2023-12-07:
I'm looking to create a very simple ble remote for the newer GoPro models - is it possible to communicate with a GoPro with Arduino based hardware? Below is the link for the hardware I am considering using. 

https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html?gad_source=1&gclid=CjwKCAiA98WrBhAYEiwA2WvhOhK_t2GmTxL6vSd8rwxYMqTErv_t_1B1E9RwJQe1_8dMcT-H29pWuhoC_vcQAvD_BwE

tcamise-gpsw, 2023-12-07:
Hello. OpenGoPro uses standard BLE communication. So as long as the Arduino supports this than yes you should be able to use it.
I can't comment on any Arduino-specific restrictions.

KonradIT, 2023-12-07:
I've used https://github.com/enclot/GoProBLE and it works pretty well

closed 2023-12-07 by tcamise-gpsw