gopro/OpenGoPro #454: Extra data returned by BLE Get Hardware Info (0x3c) command response (after MAC which is the last documented field)

pokmis, 2023-12-11:
**Component**
What is the bug in?
- Specs (BLE , HTTP)

**Describe the bug**
The response to the 'Get Hardware Info' command includes additional data no documented in the specs.

example response:
20:60:3c:00:04:00:00:00:37:0b:48:45:52:4f:39:20:42:6c:61:63:  <- initial response
80:6b:04:30:78:30:35:0f:48:44:39:2e:30:31:2e:30:31:2e:36:30:  <- continuation
81:2e:30:30:0e:43:33:34:34:31:33:32:34:38:30:34:37:39:32:0b: <- continuation
82:48:45:52:4f:39:20:42:6c:61:63:6b:0c:**64:36:33:32:36:30:62**: <- continuation
**83:35:35:31:39:35**:01:00:01:01:01:00:07:5b:22:62:6c:65:22:5d: <- continuation (extra starts here)
84:01:01: <- continuation (extra continues here)

This decodes as: 
commandSentSuccessfully:0
modelNumber: 55
lengthOfModelId: 11
modelId: "HERO9 Black"
boardType:"0x05"
firmwareVersion:"HD9.01.01.60.00"
serialNumber: "C3441324804792"
ssid: "HERO9 Black"
Mac: "**d63260b55195**"

The final sequence "01:00:01:01:01:00:07:5b:22:62:6c:65:22:5d:84:01:01" are all additional bytes not included in the documentation: https://gopro.github.io/OpenGoPro/ble_2_0#get-hardware-info

Any idea how these bytes should be interpreted?

**To Reproduce**
Steps to reproduce the behavior:
Send BLE command data [0x01, 0x3c] to GP-0072 and await the response packets on GP-0073 

**Expected behavior**
I expected the response data for Get Hardware Info to end with the MAC address

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware**
- Camera: Hero9,.
- Smartphone iPhone 12

tcamise-gpsw, 2023-12-11:
Hello. I am able to reproduce this even on the latest Hero 12 firmware. This extra data should not be getting sent from the camera so I will open a bug to fix this internally.

For now, you can safely ignore the extra data.

tcamise-gpsw, 2024-04-10:
It turns out there is no plan to change the behavior here. Instead I have [updated the documentation](https://gopro.github.io/OpenGoPro/ble/features/query.html#get-hardware-info) to note these "reserved" bytes.

closed 2024-04-10 by tcamise-gpsw