gopro/OpenGoPro #456: Hero 12 BLE cannot reconnect to camera after exiting pairing mode

mdb91, 2023-12-18:
**Component**
- GoPro Hero 12 (v2.0 and previous)
- OpenGoPro BLE Spec 2.0
- NRF52840 development board (running NRFConnect for Desktop BLE)

**Describe the bug**
I am trying to be able to reconnect to my Hero 12  via BLE when the camera is not in the dedicated pairing menu. After pairing my Hero 12, if I disconnect from the camera and leave it on, I need to reenter the pairing menu to reconnect. My application requires the ability to disconnect and reconnect to a paired camera without user intervention. With my other cameras (Hero 7 - 10) I have been able to connect to the camera and wake them up when they have been powered off for a few hours without issue. 

I am able to connect with the Quik app and "The Remote", so I know the issue is with my NRF setup and would appreciate some guidance if possible.

The issue is the Hero 12 does not respond to the connection request (CONNECT_IND) sent by the NRF52840, and the connection is terminated after six missed packets. If the Hero 12 is in the pairing screen, the connection is accepted immediately.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Hero 12 as per BLE spec instructions
2. Disconnect from Hero 12. Exit pairing menu on Hero 12
3. Scan for Hero 12 and start connection procedure
4. Issue - Hero 12 does not respond to connection request. Connection terminated.

**Expected behavior**
Packet responses from the Hero 12 confirming the connection and determining BLE settings etc.

**Screenshots**
Here is a Wireshark capture showing the connection request and following packet requests from the NRF52840. This is a previously paired device with no other devices paired to the Hero 12 or NRF52840. The Hero 12 is not in pairing mode.
[image]

Here is a capture when the Hero 12 is then put into pairing mode (connection success)
[image]

alenko, 2024-02-03:
I don’t know if you solved this, but might help to someone else, if you are using, when scanning, don't scan for GoPro in name if BT device but rather MAC address and try to connect to mac address, i had same problem as you, but i solved it with that.

mdb91, 2024-02-29:
> I don’t know if you solved this, but might help to someone else. Scanning for BT , don't scan for GoPro in name in BT device but rather MAC address and try to connect to mac address, i had same problem as you, but i solved it with that.

Hi alenko,
My issue is the Hero 12 is not completing the pairing process and not adding the device to its whitelist. I've created a new issue with more details about this (Issue #476).
I am scanning for GoPros using their MAC address once paired

tcamise-gpsw, 2024-04-11:
This is being tracked on the duplicate #476

closed 2024-04-11 by tcamise-gpsw