gopro/OpenGoPro #457: issue with Provisioning COHN on GoPro 12

Aashyk-Mohaiteen, 2023-12-20:
This issue was found while running the demo file: demos/python/sdk_wireless_camera_control/open_gopro/demos/cohn.py

I ran the demo file for provisioning and test the GoPro for COHN and ran into this issue:

[image]

Seems like the password string is not being written to the BLE characteristic and only first character is being written

**To Reproduce**
Steps to reproduce the behavior:
Run this in command prompt: python cohn.py --ssid wifi-ap --password password

**Expected behavior**
Expected the GoPro 12 to be provisioned successfully

**Hardware**
- Camera: Hero12
- Python version 3.11.4 on Windows 10

tcamise-gpsw, 2024-01-08:
That is quite an odd error.  I've only ever seen this is an anomaly in bleak.
Is this actually repeatable? If so, can you post the entire log file (i.e. gopro_demo.log)?

joelA11, 2024-01-29:
when i reduce the size of SSID, it passes. But still i get a timeout error or sometimes it just runs the LED KEEP ALIVE command

Fougl, 2024-04-04:
I get exactly the same error, any solutions yet? @joelA11  dont quite understand what you did there. @tcamise-gpsw should I send you the log file?

tcamise-gpsw, 2024-04-10:
I'm still not able to reproduce this. Also fwiw I did push a pretty major update of the SDK this week. Can someone try again and post the complete log here of the failure?

joelA11, 2024-04-22:
> I get exactly the same error, any solutions yet? @joelA11  dont quite understand what you did there. @tcamise-gpsw should I send you the log file?

Reduce the size of SSID and restrict the password to minimum characters. Also check if you've give enough ip address range in router.

The bleak module that works has some constraints with passing the ssid and password as a sequence, which has the length restriction, so it's advised to have the ssid length minimal to pass with password along.

tcamise-gpsw, 2024-04-23:
@joelA11 Can you provide some more info on this bleak limitation? Where do you see this?

Thanks

tcamise-gpsw, 2024-04-23:
I think I figured out what is going on here and it is a bug on the GoPro side. It has nothing to do with Bleak.

I can replicate the issue by sending any command that takes more than one packet (20 bytes) and uses the [general header](https://gopro.github.io/OpenGoPro/ble/protocol/data_protocol.html#general-5-bit-packets). So any command that is between 20 and 31 bytes will trigger this.
If however, I always use the [extended header](https://gopro.github.io/OpenGoPro/ble/protocol/data_protocol.html#extended-13-bit-packets) regardless of the packet size, the bug does not occur.

I'll update the Python SDK and documentation to always use extended headers.

tcamise-gpsw, 2024-04-23:
The fix has been merged and released in [v0.16.1](https://pypi.org/project/open_gopro/)

tcamise-gpsw, 2024-05-08:
Reopening becuase this also needs to be fixed in the tutorials.

closed 2024-05-09 by tcamise-gpsw