gopro/OpenGoPro #461: Python SDK BLE Protobuf: connecting to a new network doesn't work

pavlosharhan2, 2024-01-05:
Abstract:

PythonSDK - can't connecting to a new netork
MacOS
Hero10,12

Description:

I've started playing around with COHN, but stuck with connecting to network:

I am using python SDK on mac, I am not able to send RequestConnectNew message to the gopro to connect to unprovisioned network.

Connecting to already provisioned network works. For example: I connect the network using quick app, then power off- power on the gopro and send RequestConnect command using Python SDK - and it successfully connects to the network.

The error i get Is: "bleak: Device disconnected. Error writing to characteristic".

 I can't understand what is happening. Looks like the gopro is dropping BLE connection or something, and the mobile is able to handle it somehow, but Python sdk - doesn't.
 
 
I see the  same behaviour on both Hero 10 And hero 12.

The code i use:

`async def main():
    async with WirelessGoPro(enable_wifi=False) as gopro:
        print("hi")
        result = await gopro.connect_to_access_point("ssidname", "password")`

One more thing to add: I had to increase BLE Write timeout from 5 to 20 sec inside the module, otherwise even scanning didn't work for me on.

Fougl, 2024-04-04:
@pavlosharhan any solutions here? I cant even work with the demo, gives me an error #457

tcamise-gpsw, 2024-04-10:
I'm going to close this as duplicate. Let's continue discussing in #457.
FWIW that COHN demo is still working fine for me.

closed 2024-04-10 by tcamise-gpsw

Fougl, 2024-04-11:
> I'm going to close this as duplicate. Let's continue discussing in #457. FWIW that COHN demo is still working fine for me.

Yeah i managed to make it work with trial and error, will post my code inn #457. i think i dont understand exactly how it works - but i still get some errors, but the i get the cohn working in the end