gopro/OpenGoPro #462: Cohn: connect on boot

pavlosharhan2, 2024-01-05:
Is it possible to have the camera connected to a saved network in STA mode on boot automatically, without sending bluetooth commands at all?

tcamise-gpsw, 2024-01-08:
Yes that is currently the default behavior once you have COHN provisioned.

pavlosharhan2, 2024-01-08:
@tcamise-gpsw  Is it possible to have multiple networks saved so the camera Always automatically connects to the one in range, like smartphones do?

tcamise-gpsw, 2024-01-08:
That is a good question. You can definitely have multiple networks provisioned but I don't know how they are selected. I'll need to look into this.

pavlosharhan2, 2024-01-08:
@tcamise-gpsw Thank you. One more question I have regarding COHN: We want to control the camera from outside of the local Network. We do not have admin control over the network so forwarding the port is not an option. is It possible to have some tunnel between camera and our custom server to control camera from internet not forwarding the port?
I mean does it sound like it is technically possible to create such a firmware allowing us to set up this tunnel?

pavlosharhan2, 2024-01-08:
I moved the question to separate issue: https://github.com/gopro/OpenGoPro/issues/467

ksaylor, 2024-03-12:
> Yes that is currently the default behavior once you have COHN provisioned.

I've been struggling to get my HERO12 provisioned (using Python open_gopro) for a WiFi network. I've created the cert but I am not sure how to actually connect the camera to the WiFi network after that. If I run this:

`await gopro.connect_to_access_point(ssid='Linksys',password='password')`

the command never completes. CTRL-C shows that it's in an infinite loop waiting for `scan_result()` to return `proto.EnumScanning.SCANNING_SUCCESS`

`venv\Lib\site-packages\open_gopro\gopro_wireless.py", line 528, in connect_to_access_point
    if (sresult := await scan_result.get()).scanning_state != proto.EnumScanning.SCANNING_SUCCESS:`

Is there a good tutorial to walk through connecting a HERO12 to a WiFi and then controlling the camera via HTTPS? At the end of the day, I want to control multiple cameras connected to a WiFi access point. Thanks for any help you may be able to provide.