gopro/OpenGoPro #473: COHN documentation

ellis2323, 2024-01-17:
My request is a better documentation of the feature COHN.

I have a hero 11 black. I want to control my hero via an application on a smartphone to record a file and download it.
I tried OpenGOPro but I discovered that the camera create an wifi AP, so I lost the ability to send the recorded file on a website. 
I read the documentation and I suspect that the COHN is the feature I need but it's not very clear.

Is the COHN the ability for a GoPro hero 12 to connect to any WIFI LAN? What Wifi is supported ? Can I record and download a previous file at the same time ?

Best regards

Fougl, 2024-03-27:
@tcamise-gpsw i would also, like to know if I can record and download at the same time? I will definitely test on the Hero 12, but before the delivery I am a bit impatient to know. Thanks

ellis2323, 2024-04-12:
The pages are restricted. I cannot read them. Thanks

closed 2024-05-24 by github-actions[bot]