gopro/OpenGoPro #474: Gopro Remo with Hero 9-12

Wumfg, 2024-01-19:
**Is your feature request related to a problem or use case? Please describe.**
Gopro Remo is not compatible with Hero 9-12, and there are no existing remotes with haptic (vibration) feedback. 

**Describe the solution you'd like**
I would like to see if Open Gopro offers a solution to making the GoPro Remo work with the newer models, since it also uses bluetooth communication. I don't need the voice activated features, just the shutter/mode buttons and the vibration feedback about what the camera is doing. 

**Describe alternatives you've considered**
The only other alternative for this solution is creating a BLE remote myself with a microcontroller, mimicking the remo's functionality (minus the voice activation). 

**Additional context**
I'd also be open to setting up a basic wired remote with the same functionality if anyone has an existing example.

fr33z3, 2024-01-24:
Hello @Wumfg I'm working on apple watch application that gives you this functionality and a bit more. You can find it here:
https://apps.apple.com/app/id6458733064

tcamise-gpsw, 2024-04-11:
Hello. To be clear the Remote [is not supported](https://gopro.github.io/OpenGoPro/#compatibility) by Open GoPro and in any case does not use the Open GoPro interface. 

But yes you should be able to create remote yourself using the Open GoPro API if you wanted. I would recommend starting with the [tutorials](https://gopro.github.io/OpenGoPro/tutorials/).

closed 2024-04-11 by tcamise-gpsw