gopro/OpenGoPro #475: Enable narrow lens mode for streaming

CraigChisholm, 2024-01-24:
**Is your feature request related to a problem or use case? Please describe.**
When using the GoPro Quik app on GoPro Hero 10 Black it is possible to start streaming with any of four lens modes
- Narrow
- Linear
- Wide
- Superview

Using the open-gopro module in Python, the narrow view is not available as an option in `EnumLens`.

**Describe the solution you'd like**
Add `LENS_NARROW` to `EnumLens`.

**Describe alternatives you've considered**
I've tried modifying the source code but haven't yet found the correct file. I know that the value corresponding to `NARROW_LENS` should be `2`.

**Additional context**
None

tcamise-gpsw, 2024-04-11:
Yeah this looks like a bug. I'm working on updating the protobuf files to add it back in, then I'll regenerate the Python code.

tcamise-gpsw, 2024-04-11:
Actually it's possible that this option was broken. I just tested livestreaming in Quik to Hero 10 via Android and iOS and I do not see an option to even set then lens. Where are you seeing this?

tcamise-gpsw, 2024-04-12:
It turns out that I had already discovered this in the past and have a bug open to fix the documentation: #374 

So I will close this as duplicate and update the documentation in that linked ticket.

closed 2024-04-12 by tcamise-gpsw

CraigChisholm, 2024-04-13:
I don't think this duplication mark is correct. I am using a Hero 10 _Black_ and I can set the field of view to Linear, Wide, and Superview before starting the stream and I do see the obvious change in the image. My problem is that when I am using the camera touch screen it has an additional Narrow mode which is not available in the Protobuf commands. Which is different to the bug #374 where certain cameras do not support setting the field of view.

tcamise-gpsw, 2024-04-16:
From an official usage perspective: the only FOV's that are supported for livestreaming are those dynamically reported from the `live_stream_lens_supported_array` in [NotifyLiveStreamStatus](https://gopro.github.io/OpenGoPro/ble/protocol/protobuf.html#proto-notifylivestreamstatus)). That is, the only way to configure FOV for livestream is to use one of these reported options and set the `window_size` parameter of [RequestSetLiveStreamMode](https://gopro.github.io/OpenGoPro/ble/protocol/protobuf.html#requestsetlivestreammode). 

Hero 10 Black does not set this `live_stream_lens_supported_array` so this parameter is therefore not configurable. 

The UI Narrow mode that you are referring to is the Lens setting for one of the Photo, Video, or Timelapse Lens [Settings](https://gopro.github.io/OpenGoPro/ble/features/settings.html#lens-121). This has no relevance on livestreaming and would get overridden by the livestreaming parameter if it did exist. Since this is not configurable on Hero 10 Black, it gets overridden by the default.

This is the same reason the Quik App does not provide a lens option for livestreaming with Hero 10 Black.