gopro/OpenGoPro #476: GoPro12 Paired Device not added to Whitelist

mdb91, 2024-01-28:
**Components:**
- GoPro 12 (Version 02.00)
- Python Tutorials (Windows 10, Python 3.10.11)
- Custom BLE PCB (based off of C and Python SDK) using NRF52840 MCU

**Description:**
After I pair the GoPro12 with either the Windows PC running the Python tutorials, or my custom PCB project; I am unable to reconnect to the GoPro without putting the camera back into the pairing mode using the UI. I'm using Wireshark to see the packets, and when I try to reconnect (when not in camera pairing mode), the camera doesn't respond to the initial CONNECT_IND request from the client. After six timeouts, the client disconnects from the camera (as per BLE standard).

My question is what am I missing when I am pairing to the GoPro12? I don't think my client is being added to the camera's whitelist, but I am following the procedure listed in the BLE spec (connect, pair, discover, enable notifications).

When I query the Pairing State status (ID 19) of the camera after finishing pairing, I get a response of 0x01 (started), not 0x04 (completed). I modified Tutorial 4 BLE Queries - Poll Multiple Settings Value to request the Pairing State from the camera and received the same 0x01 (started) response [see screenshots].

I have had no issues with the other cameras I am testing (GoPro 9 and 10).

When I am pairing, the GoPro has the confirmation beep and I have checked it advertises correctly after a power cycle (using LightBlue and NRFConnect).

**Supporting Info:**
Python Tutorial Screenshots and Wireshark Trace:
[GoPro BLE Python.zip](https://github.com/gopro/OpenGoPro/files/14074988/GoPro.BLE.Python.zip)
- "Python_GoPro12_BLE_Connect_Pair_and_reconnect_attempt.PNG" - screenshot of Python Tutorial 1. Pairing was completed and then UI was exited and a new connection was attempted but failed.
- "Python_GoPro12_BLE_Connect_sniffer_trace.pcapng" - Wireshark trace of Tutorial 1
- "Python_GoPro12_Pair_State_Query.PNG" - screenshot of Tutorial 4 (modified to request Pairing State status) showing the pairing state is "Started".

NRF52 Wireshark Traces:
[GoPro BLE NRF52.zip](https://github.com/gopro/OpenGoPro/files/14074993/GoPro.BLE.NRF52.zip)
- "240128 - GoPro12_NRF52_Pairing_w_notification.pcapng" - Wireshark trace of pairing procedure with GoPro12 and NRF52. Note query for Pairing State at end shows status is "Started".
- 240128 - GoPro12_NRF52_no_connection_response.pcapng" - Wireshark trace of NRF52 attempting to reconnect with GoPro12 after pairing and power cycling. The GoPro12 was not put into pairing mode.

mdb91, 2024-03-15:
As an update to this issue, I have successfully tested my NRF52840 project with a Hero 11 Black which worked as expected. This means the pairing issue is only with the Hero 12 cameras.

tcamise-gpsw, 2024-04-11:
This does seem like a bug on the GoPro and is being investigated internally.