gopro/OpenGoPro #478: OpenAPI specification does not pass validation

axeljaeger, 2024-02-04:
**Component**
What is the bug in?
- HTTP spec

**Describe the bug**
The provided openapi.json contains some errors so they do not pass validation. The errors are:

```
Swagger schema validation failed.
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must have required property 'required'
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "then" schema
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "else" schema

Error: Swagger schema validation failed.
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must have required property 'required'
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "then" schema
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "else" schema

    at o (https://apitools.dev/swagger-parser/online/js/bundle.min.js:1:95564)
    at https://apitools.dev/swagger-parser/online/js/bundle.min.js:29:45042

SyntaxError: Swagger schema validation failed.
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must have required property 'required'
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "then" schema
  #/paths/~1gopro~1camera~1shutter~1{mode}/get/parameters/0 must match "else" schema

    at Function.o [as syntax] (https://apitools.dev/swagger-parser/online/js/bundle.min.js:1:95564)
    at validateSchema (https://apitools.dev/swagger-parser/online/js/bundle.min.js:1:6322)
    at SwaggerParser.validate (https://apitools.dev/swagger-parser/online/js/bundle.min.js:1:3391)
```
This is causing problem when creating code from the specification. If you are using [openapi-generator](https://github.com/OpenAPITools/openapi-generator), you have to pass `--skip-validate-spec` to make the code generation start.

**To Reproduce**
Paste the spec to https://apitools.dev/swagger-parser/online/ and press validate

**Additional context**
Hey guys, I am so happy that you moved to OpenAPI compatible spec. Thank you so much! Now I am challenging it by throwing the complete OpenAPI ecosystem on it.

tcamise-gpsw, 2024-04-10:
This is interesting. The spec is validated by [Redocly ](https://redocly.com/redocly-cli/). 
My guess here is that I have set a more lax set of rules for validation than what is being used by Swagger.

Is this wrong? Perhaps. I'll need to look into this some more.

tcamise-gpsw, 2024-04-12:
It turns out i had disabled Redocly verification. I turned this back on and fixed some errors but there are still some errors that show from openapi-generator and not in Redocly.  I guess I'll add both of this to the CI flow (and maybe Swagger validate also in the future).

I should have the doubly-validated spec out next week.

tcamise-gpsw, 2024-04-12:
This is fixed as of [this commit](https://github.com/gopro/OpenGoPro/commit/936212f79d68a39e84d0ad8fde673d41d7607501) and should remain so since we are now validating after building the spec.

I did not add in Swagger validation since hopefully everything is covered by Redocly and openapi-generator validation but can if needed in the future.

closed 2024-04-12 by tcamise-gpsw