gopro/OpenGoPro #485: I can not find any command or setting on how to turn on/off GPS.

fr33z3, 2024-02-15:
I can not find in documentation how could I send a command to turn on/off GPS. Could you provide that in documentation?

bfreeland-gpsw, 2024-03-13:
This isn't currently documented so I'll provide the endpoints below. We plan on expanding the OGP settings documentation in future to include all supported settings including GPS.

Enable GPS:
`http://10.5.5.9:8080/gopro/camera/setting/83/1`

Disable GPS:
`http://10.5.5.9:8080/gopro/camera/setting/83/0`

tcamise-gpsw, 2024-03-13:
The docs have been updated now and you can find the setting at in [BLE](https://gopro.github.io/OpenGoPro/ble/features/settings.html#gps-83) and [HTTP](https://gopro.github.io/OpenGoPro/http#tag/settings/operation/GPCAMERA_CHANGE_SETTING::83)

closed 2024-03-13 by tcamise-gpsw

fr33z3, 2024-04-04:
Thank you so much @tcamise-gpsw