gopro/OpenGoPro #487: WirelessGoPro._periodic_keep_alive coroutine is left running if connection failed.

epheo, 2024-02-17:
**Component**
In the Python SDK.

**Describe the bug**
If the connection to a GoPro fails, the WirelessGoPro._periodic_keep_alive coroutine is left open.

**To Reproduce**
Start connection to a GoPro and have it fail (stop ble adapter or remove GoPro battery)
Then we can observe _periodic_keep_alive by checking the running coroutines:

tasks = asyncio.all_tasks()
coroutine_name = task.get_coro().__qualname__

**Expected behavior**
I would expect the keepalive to be stopped once the Connection fails.

**Hardware**
Hero 11
Linux 6

**Additional context**
I'll try to investigate later if this can be solved by initiating connection within a contextlib.AsyncExitStack().

tcamise-gpsw, 2024-04-10:
I've made some changes to this in the last release. However the bigger issue here is that the Wire(less)GoPro objects are not designed to persist through subsequent connections. I haven't spent much time on this because I don't think bleak robustly manages, at least on Windows, failed / reconnections.

In any case, there is a large investigation required to improve the robustness here.

tcamise-gpsw, 2024-07-01:
Another symptom: BLE connection drops after provisioning COHN even though there is still a communication channel through COHN cause state management to break. We're going to need to rearchitect the entire connection scheme to be scalable for multiple connections / types.

tcamise-gpsw, 2024-08-30:
Relevent bleak ticket here: https://github.com/hbldh/bleak/issues/367#issuecomment-2306384359

shiratama820, 2024-10-18:
@tcamise-gpsw 
I came from the #555 thread.
I am also experiencing the same issue as in #555.
I am using Python 3.9 with the HERO13 Black.

Is there any prospect of this issue being resolved?
Could you please provide an update on the current status?

Best regards,
shiratama820

MattL-NZ, 2025-03-14:
Is there any update on this ticket? It's causing us quite a bit of pain communicating via the python SDK with the Hero 13

tcamise-gpsw, 2025-03-14:
No and it needs a deep investigation into various bleak errors on all OS's. 
Can you post the errors you are seeing?

MattL-NZ, 2025-03-14:
@tcamise-gpsw I get really varying results. These are logs from my Raspberry Pi trying to connect two cameras, one after the other.

2025-03-14 19:25:22,930 - EpicGoPro - INFO - Setting up GoPro cameras...
Failed to find a device in 5 seconds. Retrying #1
2025-03-14 19:25:25,791 - EpicGoPro - INFO - GoPro E1:24:8C:EC:22:DF: GoPro 0532 is connected via BLE
Failed to find a device in 5 seconds. Retrying #1
Failed to find a device in 5 seconds. Retrying #2
Failed to find a device in 5 seconds. Retrying #3
2025-03-14 19:25:29,061 - EpicGoPro - INFO - GoPro C0:10:30:52:ED:BD: GoPro 7983 is connected via BLE

Sometimes it connects straight away, other times it refuses to connect and jumps through all the retry methods almost instantly, even setting the timeout to 30 seconds, there is no 30 second break between checks, or not according to the logs.

Other times only one camera will connect, and one won't. 
Is there a way to provide you better logs? 

```
async def setup_gopro(self, gopros: List[GoProDeviceModel]) -> list[WirelessGoPro]:
        """Set up multiple GoPro cameras with sequential operations."""
        self.logger.info("Setting up GoPro cameras...")
        connected_gopros = []

        gopro: WirelessGoPro | None = None
        for gopro in gopros:
            gopro_name = self.get_gopro_name(gopro.serialNumber)
            try:
                gopro = WirelessGoPro(gopro_name, enable_wifi=False)
                await gopro.open(timeout=5, retries=5)
                if gopro.is_ble_connected:
                    self.logger.info(f"GoPro {gopro.identifier} is connected via BLE")
                    assert await gopro.ble_command.set_camera_control(
                        camera_control_status=proto.EnumCameraControlStatus.CAMERA_EXTERNAL_CONTROL
                    )
                    connected_gopros.append(gopro)
                else:
                    self.logger.error(f"GoPro {gopro.identifier} is not connected via BLE")
            except Exception as e:
                self.logger.error(f"Error setting up GoPro: {e}")

        return connected_gopros
```

tcamise-gpsw, 2025-03-14:
The logs are quite detailed already. But it sounds like you're only looking at the CLI log. There should be an extremely detailed `.log` file next to wherever you are calling the script from. So let's start with those.

MattL-NZ, 2025-03-14:
@tcamise-gpsw here are the logs for trying to connect to two devices on my macbook where only one successfully connected.

```
INFO     09:34:21.610 Scanning for GoPro 0532 bluetooth devices...                                                                                                                                                                                                                bleak_wrapper.py:121
INFO     09:34:24.190    Discovered: 2952AF0B-53BD-C556-64F9-D95DAF0069E7: GoPro 7983                                                                                                                                                                                             bleak_wrapper.py:134
WARNING  09:34:24.199 Failed to find a device in 5 seconds. Retrying #1                                                                                                                                                                                                                   client.py:86
INFO     09:34:24.203 Scanning for GoPro 0532 bluetooth devices...                                                                                                                                                                                                                bleak_wrapper.py:121
INFO     09:34:24.308    Discovered: 0C8C3694-332B-9A80-7103-9AD3CE3F3B0C: GoPro 0532                                                                                                                                                                                             bleak_wrapper.py:134
INFO     09:34:24.312 Found 1 matching devices.                                                                                                                                                                                                                                   bleak_wrapper.py:147
INFO     09:34:24.314 Establishing the BLE connection                                                                                                                                                                                                                                    client.py:107
INFO     09:34:24.316 Establishing BLE connection to 0C8C3694-332B-9A80-7103-9AD3CE3F3B0C: GoPro 0532...                                                                                                                                                                          bleak_wrapper.py:214
INFO     09:34:25.247 Discovering characteristics...                                                                                                                                                                                                                              bleak_wrapper.py:359
INFO     09:34:25.842 Done discovering characteristics!                                                                                                                                                                                                                           bleak_wrapper.py:404
INFO     09:34:25.847 Enabling notifications...                                                                                                                                                                                                                                   bleak_wrapper.py:322
INFO     09:34:26.377 Done enabling notifications                                                                                                                                                                                                                                 bleak_wrapper.py:328
INFO     09:34:26.384                                                                                                                                                                                                                                                            gopro_wireless.py:726
         <<<<<<<<--------------                                                                                                                                                                                                                                                                       
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                                                                                                                                                                                                                                               
             "status_id" : "StatusId.ENCODING",                                                                                                                                                                                                                                                       
             "protocol" : "Protocol.BLE",                                                                                                                                                                                                                                                             
             "uuid" : "Query",                                                                                                                                                                                                                                                                        
[... cut ...]
ERROR    09:34:29.001 Error while opening: A scan timed out without finding a device                                                                                                                                                                                             gopro_wireless.py:325
2025-03-15 09:34:29,005 - EpicGoPro - ERROR - Error setting up GoPro: A scan timed out without finding a device
ERROR    09:34:29.005 FailedToFindDevice('A scan timed out without finding a device')     
```

MattL-NZ, 2025-03-14:
@tcamise-gpsw Here are the logs for my Raspberry PI where the devices both connected fine but failed to connect via COHN/WiFi

```
2025-03-14 20:38:51,268 - EpicGoPro - INFO - Setting up GoPro cameras...
INFO     20:38:51.275 Scanning for GoPro 0532 bluetooth     bleak_wrapper.py:121
         devices...                                                             
INFO     20:38:51.387    Discovered: CD:71:8A:99:00:AD:     bleak_wrapper.py:134
         GoPro 7983                                                             
WARNING  20:38:51.392 Failed to find a device in 5 seconds.         client.py:86
         Retrying #1                                                            
INFO     20:38:51.394 Scanning for GoPro 0532 bluetooth     bleak_wrapper.py:121
         devices...                                                             
INFO     20:38:51.544    Discovered: CD:71:8A:99:00:AD:     bleak_wrapper.py:134
         GoPro 7983                                                             
WARNING  20:38:51.549 Failed to find a device in 5 seconds.         client.py:86
         Retrying #2                                                            
INFO     20:38:51.550 Scanning for GoPro 0532 bluetooth     bleak_wrapper.py:121
         devices...                                                             
INFO     20:38:51.578    Discovered: C6:37:2E:00:CD:31:     bleak_wrapper.py:134
         GoPro 0532                                                             
INFO     20:38:51.582 Found 1 matching devices.             bleak_wrapper.py:147
INFO     20:38:51.583 Establishing the BLE connection              client.py:107
INFO     20:38:51.584 Establishing BLE connection to        bleak_wrapper.py:214
         C6:37:2E:00:CD:31: GoPro 0532...                                       
INFO     20:38:53.772 Pairing with bluetoothctl             bleak_wrapper.py:263
INFO     20:38:54.003 Discovering characteristics...        bleak_wrapper.py:359
INFO     20:38:55.027 Done discovering characteristics!     bleak_wrapper.py:404
INFO     20:38:55.029 Enabling notifications...             bleak_wrapper.py:322
INFO     20:38:55.953 Done enabling notifications           bleak_wrapper.py:328
INFO     20:38:55.955                                      gopro_wireless.py:723
         <<<<<<<<--------------                                                 
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                         
             "status_id" : "StatusId.ENCODING",                                 
             "protocol" : "Protocol.BLE",                                       
             "uuid" : "Query",                                                  
         <<<<<<<<--------------                                                 
                                                                                
INFO     20:38:56.100                                      gopro_wireless.py:656
         -------------->>>>>>>>                                                 
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                         
             "status" : "ErrorCode.SUCCESS",                                    
             "protocol" : "Protocol.BLE",                                       
             "data" : {                                                         
                 "StatusId.ENCODING" : "False",                                 
             },                                                                 
         -------------->>>>>>>>                                                 
                                                                                
INFO     20:38:56.103                                      gopro_wireless.py:723
         <<<<<<<<--------------                                                 
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                         
             "status_id" : "StatusId.BUSY",                                     
             "protocol" : "Protocol.BLE",                                       
             "uuid" : "Query",                                                  
         <<<<<<<<--------------                                                 
                                                                                
INFO     20:38:56.197                                      gopro_wireless.py:656
         -------------->>>>>>>>                                                 
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                         
             "status" : "ErrorCode.SUCCESS",                                    
             "protocol" : "Protocol.BLE",                                       
             "data" : {                                                         
                 "StatusId.BUSY" : "False",                                     
             },                                                                 
         -------------->>>>>>>>                                                 
                                                                                
INFO     20:38:56.200 BLE is ready!                        gopro_wireless.py:596
INFO     20:38:56.224                                      gopro_wireless.py:723
         <<<<<<<<--------------                                                 
             "id" : "CmdId.SET_DATE_TIME_DST",                                  
[... cut ...]
             "status" : "ErrorCode.SUCCESS",                                    
             "protocol" : "Protocol.BLE",                                       
             "data" : {                                                         
                 "result" :                                                     
         "EnumResultGeneric.RESULT_SUCCESS",                                    
             },                                                                 
         -------------->>>>>>>>                                                 

```