gopro/OpenGoPro #488: Error using gopro-photo and gopro-video demos with Gopro 9

acechamp, 2024-02-18:
**Component**
What is the bug in?
- Demos (python sdk)
- Specs BLE , HTTP(USB)

**Describe the bug**
gopro-photo and gopro-video do not function. however gopro-webcam does function
tested on two Gopro 9's and two different windows machines to reduce variables

I've tried
-installing older versions on opengopro(per an old issue report) did not help just caused more errors probably due to incompatible packages and py version
-wired functionality, and BLE functionality(come up with different errors but seem to both stop after the shutter on command is sent to the gopro) see screenshots and log
-searching for similar issues here and online(Nothing matched this issue)
-whispering "please just work" a few times(did not work)

**To Reproduce**
Steps to reproduce the behavior:
Install Python 3.10.11(tried 3.11.8 first then downgraded)
run       pip install open-gopro
run       gopro-photo or gopro-video 2
Gopro 9 BLE connects, then WIFI shows that it connects
Camera then correctly switches to photo or video mode 
but photo is not produced and a few errors come up.

BLE SCREENSHOTS
[image]
[image]
[image]

USB SCREENSHOTS
[image]
[image]
[image]
[image]

ERROR LOG FROM gopro-photo --log
[LOG.txt](https://github.com/gopro/OpenGoPro/files/14321547/LOG.txt)

**Hardware**
- Camera: Hero9 x2(both same issue)
- Windows x2 machines (both have same issue)

**Additional context**
I am not a programmer but I really tried everything I can think of to make this work(And read through the relevant tutorials and documentation). I'm building for underwater time-lapse/livestream enclosure and I bought both gopros specifically for this project because this API was available for them. I spent 10 hours trying to make this work so far and just haven't gotten anywhere. I hope this can be fixed if its a bug, Thanks.

acechamp, 2024-03-10:
Update, just bought a GoPro 11 to test with and it worked out of the box. Defiantly is a bug with the 9 and the demo code. maybe it was updated for the newer cameras and it broke compatibility with the 9? but then I would think that older code bases would work, I've seen people report getting the 9's to work with the test code before so at one point it was working.

tcamise-gpsw, 2024-04-10:
Hello. The demos are only tested against the latest cameras and it is up to the user to handle per-camera functionality. 

For the example API above, you can see it is marked as not supported in Hero 9 in the HTTP [spec](https://gopro.github.io/OpenGoPro/http#tag/Control/operation/OGP_SET_WIRED_USB_CONTROL):

[image]

closed 2024-04-10 by tcamise-gpsw