gopro/OpenGoPro #504: HTTP API => Internal Server Error

Ektacom, 2024-03-29:
When controlling the USB camera through the HTTP API through USB, sometimes the answer to a HTTP request is "Internal Server Error" with a HTTP status code equal to 500. 
When this response appears, then the gopro HTTP server is blocked: none request can be done to the gopro and there is no way to recover. The only solution is to unplug the camera.
I've experienced this problem on Gopro Black 12 (I don't know if the problem exit on other gopro cameras). Unfortunately I'm not able to reproduce the problem but from my experience, it can appear randomly on any request or request sequence.
- What could be the reason of this message ? Is there a way to mitigate this problem ?
- Is there a way to recover when it appears whithout having to restart the gopro camera ?

pavlosharhan2, 2024-03-29:
I have a hope that this is fixed in today's 2.20 firmware!

pavlosharhan2, 2024-03-29:
@Ektacom 

Hey, having the same issue, going to try today's new 2.20 firmware. Will see how it goes.

By the way, are you sure that unplugging the camera helps with this? All the people known to me having this problem say that only resetting the camera helps ( sometimes )

Ektacom, 2024-04-02:
@pavlosharhan2 
I don't have any battery in the camera. The camera is powered throuyh USB : this may explain why when I unplug USB cable it "solves" my problem.

pavlosharhan2, 2024-04-04:
UPD:
Haven't noticed this bug in the latest 2.20 Labs firmware

tcamise-gpsw, 2024-07-17:
I'm unable to reproduce this with the latest firmware. Is anyone still having this issue?

aoyurs, 2024-09-04:
I encountered the same issue with firmware version v2.20.70.

closed 2024-10-15 by github-actions[bot]