gopro/OpenGoPro #505: Not able to interract with BLE API via protobuf

h3lgi, 2024-04-01:
**Component**
BLE API + Protobuf

**Describe the bug**
I am implementing a master device that has to configure and start the  `RTMP` streaming on the `GoPro Hero9`. The device is based on `esp32s3` and I cannot use the GoPro access point. 

By following the protocol description I implemented the protocol itself and some basic commands. 
All TLV commands are working as expected which is not the case with the protobuf once. When the command or response is bigger than 20 bytes I am not receiving anything or receiving the `RESULT_ILL_FORMED`

I rechecked the packetization a few times and it follows the [rules](https://gopro.github.io/OpenGoPro/ble/protocol/data_protocol.html#packet-headers) in the protocol description:
```[ header with  <5/13/16 bitmask> and <message length> ] [ <feature ID> ] [ <Action ID> ] [ <data> ]```
if it is longer than 20 bytes.
```[ <continuation header counting from 0 > and <continuation mask>]  [ <data> ]```

**To Reproduce**
Steps to reproduce the behavior with scan for access points command
1. Connect and pair with the camera 
2. Subscribe to all readable characteristics
3. Send command to start scanning: `0x2 0x2 0x2` 
4. Await start scan response and receiving: `0x6 0x2 0x82 0x8 0x1 0x10 0x2` which is decoded to: 
``` JSON
{
  "result":  1,
  "scanning_state": 2
}
```
5. Waiting for start scanning notification and receiving: `0xa 0x2 0xb 0x8 0x5 0x10 0x2 0x18 0xf 0x20 0x0` which is decoded to:
``` JSON
{
  "scanning_state": 5,
  "scan_id": 2,
  "total_entries": 15,
  "total_configured_ssid": 0
}
```
6. Sending request to request entries: `0x8 0x2 0x4 0x8 0x0 0x10 0xf 0x18 0x2` encoded from: 
``` JSON
{
  "start_index": 0,
  "max_entries": 15,
  "scan_id": 2
}
```
7. Waiting response and receiving: `0x8 0x2 0x84 0x8 0x2 0x10 0x0 0x18 0x0` which cannot be decoded to `ResponseGetApEntries` protobuf message.

Similar behavior with the other `protobuf` formated commands: 
- `RequestConnectNew `: no response or notification
- `RequestSetLiveStreamMode`: no response or notification

**Expected behavior**
Receiving decodable response or reaction to protobuf commands over BLE

**Hardware**
- esp32 S3: as remote control device
- nanoPb: as protobuf serializer
- BLE: as protocol 
- GoPro Hero9 Black
- Camera firmware version:  v1.72

h3lgi, 2024-04-03:
Well, it was partially related to issue #494  and the wrong endian in the bit masks on my package formation algorithm.

closed 2024-04-03 by h3lgi

alexkok, 2025-11-07:
Hey @h3lgi were you able to get the RequestConnectNew packet working? I've many commands working, but with this is one, and the RequestSetLivestreamMode I'm facing that the GoPro not even responding, or sending a notification when I send this one. Even RequestConnect works when I added the AP via qr code before. The write goes OK seems like, just no response at all, as if it doesn't know about the command.

So wondering if you managed to ge the RequestConnectNew/RequestSetLivestreamMode working on the Hero9, or whether this is only supported in higher versions. I'm using fw v1.72.70.