gopro/OpenGoPro #506: Apple has officially retired the airport command-line utility on macOS Sonoma 14.4

epheo, 2024-04-04:
**Component**
Python SDK

**Description**
SSID Scan does not work anymore on recent Mac OS X version.

**To Reproduce**
Try connecting to a GoPro using Mac OS Sonoma 14.4

**Expected behavior**
Python SDK should migrate to wdutil as per Apple recomandation.

**Screenshots**

`
~ % airport
WARNING: The airport command line tool is deprecated and will be removed in a future release.
For diagnosing Wi-Fi related issues, use the Wireless Diagnostics app or wdutil command line tool.
`

epheo, 2024-04-05:
So `wdutil` only returns the known SSIDs...
I'm struggling to find a suitable alternative to `airport -scan`.

One (not so straightforward) alternative could be to conditionally add pyobjc dependency if system is darwin (can rely on sys_platform marker https://python-poetry.org/docs/dependency-specification/#using-environment-markers) and make use of the native ObjC `/System/Library/Frameworks/CoreWLAN.framework`.

```
[tool.poetry.dependencies]
pyobjc = {version = "*", markers = "sys_platform == 'darwin'"}
```

```
import objc
from CoreWLAN import *

def scan_wifi_networks(interface_name=None):
    # Load CoreWLAN framework
    objc.loadBundle('CoreWLAN',
                    bundle_path='/System/Library/Frameworks/CoreWLAN.framework',
                    module_globals=globals())

    # Create a CWInterface object
    if interface_name:
        interface = CWInterface.interfaceWithName_(interface_name)
    else:
        interface = CWInterface.interface()

    # Scan for networks
    error = None
    networks, error = interface.scanForNetworksWithSSID_error_(None, None)
    
    if error:
        print(f"Error scanning for Wi-Fi networks: {error}")
    else:
        for network in networks:
            print(f"SSID: {network.ssid()}")
```

sfoley-gpqa, 2024-04-05:
> So `wdutil` only returns the known SSIDs... I'm struggling to find a suitable alternative to `airport -scan`.
> 
> One (not so straightforward) alternative could be to conditionally add pyobjc dependency if system is darwin (can rely on sys_platform marker https://python-poetry.org/docs/dependency-specification/#using-environment-markers) and make use of the native ObjC `/System/Library/Frameworks/CoreWLAN.framework`.
> 
> ```
> [tool.poetry.dependencies]
> pyobjc = {version = "*", markers = "sys_platform == 'darwin'"}
> ```
> 
> Then something like:
> 
> ```
> import objc
> from CoreWLAN import *
> 
> def scan_wifi_networks(interface_name=None):
>     # Load CoreWLAN framework
>     objc.loadBundle('CoreWLAN',
>                     bundle_path='/System/Library/Frameworks/CoreWLAN.framework',
>                     module_globals=globals())
> 
>     # Create a CWInterface object
>     if interface_name:
>         interface = CWInterface.interfaceWithName_(interface_name)
>     else:
>         interface = CWInterface.interface()
> 
>     # Scan for networks
>     error = None
>     networks, error = interface.scanForNetworksWithSSID_error_(None, None)
>     
>     if error:
>         print(f"Error scanning for Wi-Fi networks: {error}")
>     else:
>         for network in networks:
>             print(f"SSID: {network.ssid()}")
> ```

This seems like a good solution; however, on my system, the SSID is `None` for every `network` object. I've searched for a solution, gave Full Disk Access to both iTerm and terminal but no luck so far. I also tried to grant Location Services access to iTerm/terminal but I don't see an option to manually add them in macOS 14.3.1.

If you have further input, please feel free to offer it! 😬

epheo, 2024-04-06:
I do not own a Macintosh, and MacOS Ventura is the latest I can virtualize atm.
After borrowing a system with 14.4.1 for a moment I could get a correct SSID as follow:

```
[...]
>>> import CoreLocation
>>> location_manager = CoreLocation.CLLocationManager.alloc().init()
>>> location_manager.startUpdatingLocation()
>>> networks, error = iface.scanForNetworksWithName_error_( None, None )
>>> print(networks)
```

Dependencies:
- pyobjc-framework-CoreWLAN
- pyobjc-framework-CoreLocation

I tested on MacOS 14.4.1 with Python 3.11 installed via brew and pyobjc 10.2.
The following references seems to indicate that behavior may differ between systems and Python Versions.
- https://github.com/ronaldoussoren/pyobjc/issues/600
- https://forums.developer.apple.com/forums/thread/748518

sfoley-gpqa, 2024-04-08:
> I do not own a Macintosh, and MacOS Ventura is the latest I can virtualize atm. After borrowing a system with 14.4.1 for a moment I could get a correct SSID as follow:
> 
> ```
> [...]
> >>> import CoreLocation
> >>> location_manager = CoreLocation.CLLocationManager.alloc().init()
> >>> location_manager.startUpdatingLocation()
> >>> networks, error = iface.scanForNetworksWithName_error_( None, None )
> >>> print(networks)
> ```
> 
> Dependencies:
> 
> * pyobjc-framework-CoreWLAN
> * pyobjc-framework-CoreLocation
> 
> I tested on MacOS 14.4.1 with Python 3.11 installed via brew and pyobjc 10.2. The following references seems to indicate that behavior may differ between systems and Python Versions.
> 
> * [CoreWLAN returning None for SSID/BSSID ronaldoussoren/pyobjc#600](https://github.com/ronaldoussoren/pyobjc/issues/600)
> * https://forums.developer.apple.com/forums/thread/748518

Running macOS Sonoma 14.3.1 (23D60) and using the code below, I'm getting `null` for all SSIDs:

```shell
python3 -m venv /tmp/scantest
/tmp/scantest/bin/python3
Python 3.10.13 (main, Sep 13 2023, 13:03:08) [Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
```

```python
import CoreLocation
from CoreWLAN import CWInterface, CWWiFiClient

wifi_client: CWWiFiClient = CWWiFiClient.sharedWiFiClient()
interface = wifi_client.interface()
location_manager = CoreLocation.CLLocationManager.alloc().init()
location_manager.startUpdatingLocation()
networks, error = interface.scanForNetworksWithName_error_( None, None )
print(networks)
```

epheo, 2024-04-08:
weird... 
So I tried combining both Python 3.11 (from brew) and 3.9 (from system), PyObjc 9.2 (to satisfy Bleak dependencies) and 10.2 (latest) and all seems to be working fine on 14.4.1
Wondering if 14.3 and 14.4 are handling this differently (if yes, wow... such mess).

```
% poetry show |egrep "(corewlan|corelocation)" 
pyobjc-framework-corelocation  9.2            Wrappers for the framework Co...
pyobjc-framework-corewlan      9.2            Wrappers for the framework Co...

% poetry run python3 --version
Python 3.11.8

% sw_vers
ProductName:		macOS
ProductVersion:		14.4.1
BuildVersion:		23E224

% poetry run python3
Python 3.11.8 (main, Feb  6 2024, 21:21:21) [Clang 15.0.0 (clang-1500.1.0.2.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import CoreLocation
>>> from CoreWLAN import CWInterface, CWWiFiClient
>>> wifi_client: CWWiFiClient = CWWiFiClient.sharedWiFiClient()
>>> interface = wifi_client.interface()
>>> location_manager = CoreLocation.CLLocationManager.alloc().init()
>>> location_manager.startUpdatingLocation()
>>> networks, error = interface.scanForNetworksWithName_error_( None, None )
>>> print(networks)
{(
    <CWNetwork: 0x60000289c620> [ssid=Almacen Yuniko, bssid=<redacted>, security=WPA2 Personal, rssi=-88, channel=<CWChannel: 0x60000289cb20> [channelNumber=10(2GHz), channelWidth={40MHz(-1)}], ibss=0],
    <CWNetwork: 0x60000289c6a0> [ssid=lar 2, bssid=<redacted>, security=WPA/WPA2 Personal, rssi=-87, channel=<CWChannel: 0x60000289cb50> [channelNumber=7(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x60000289c780> [ssid=DIR-615-0914, bssid=<redacted>, security=WPA2 Personal, rssi=-87, channel=<CWChannel: 0x60000289cb80> [channelNumber=13(2GHz), channelWidth={40MHz(-1)}], ibss=0],
    <CWNetwork: 0x60000289c690> [ssid=LAR1, bssid=<redacted>, security=WPA2 Personal, rssi=-50, channel=<CWChannel: 0x60000289cbb0> [channelNumber=2(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x60000289c860> [ssid=Lar13, bssid=<redacted>, security=WPA/WPA2 Personal, rssi=-74, channel=<CWChannel: 0x60000289cbe0> [channelNumber=1(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x60000289c920> [ssid=NETGEAR71, bssid=<redacted>, security=WPA2 Personal, rssi=-91, channel=<CWChannel: 0x60000289cc10> [channelNumber=9(2GHz), channelWidth={20MHz}], ibss=0],
    <CWNetwork: 0x60000289c9b0> [ssid=Tara, bssid=<redacted>, security=WPA/WPA2 Personal, rssi=-94, channel=<CWChannel: 0x60000289cc40> [channelNumber=6(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x60000289ca90> [ssid=Administración, bssid=<redacted>, security=WPA2 Personal, rssi=-84, channel=<CWChannel: 0x60000289cc70> [channelNumber=8(2GHz), channelWidth={20MHz}], ibss=0]
)}
```

```
% sw_vers
ProductName:		macOS
ProductVersion:		14.4.1
BuildVersion:		23E224

% python3 -m venv /tmp/scantest
% /tmp/scantest/bin/pip3 install pyobjc

% /tmp/scantest/bin/pip3 list |egrep "(CoreWLAN|CoreLocation)" 
pyobjc-framework-CoreLocation                     10.2
pyobjc-framework-CoreWLAN                         10.2

% /tmp/scantest/bin/python3
Python 3.9.6 (default, Feb  3 2024, 15:58:28) 
[Clang 15.0.0 (clang-1500.3.9.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import CoreLocation
>>> from CoreWLAN import CWInterface, CWWiFiClient
>>> wifi_client: CWWiFiClient = CWWiFiClient.sharedWiFiClient()
>>> interface = wifi_client.interface()
>>> location_manager = CoreLocation.CLLocationManager.alloc().init()
>>> location_manager.startUpdatingLocation()
>>> networks, error = interface.scanForNetworksWithName_error_( None, None )
>>> print(networks)
{(
    <CWNetwork: 0x600002080200> [ssid=lar 2, bssid=(null), security=WPA/WPA2 Personal, rssi=-84, channel=<CWChannel: 0x600002088000> [channelNumber=7(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x6000020802d0> [ssid=LAR14, bssid=(null), security=WPA/WPA2 Personal, rssi=-91, channel=<CWChannel: 0x600002095300> [channelNumber=5(2GHz), channelWidth={40MHz(-1)}], ibss=0],
    <CWNetwork: 0x600002080380> [ssid=ETAZLA, bssid=(null), security=WPA2 Personal, rssi=-85, channel=<CWChannel: 0x600002095330> [channelNumber=4(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x6000020802c0> [ssid=LAR1, bssid=(null), security=WPA2 Personal, rssi=-55, channel=<CWChannel: 0x600002095360> [channelNumber=2(2GHz), channelWidth={40MHz(+1)}], ibss=0],
    <CWNetwork: 0x600002080460> [ssid=DIR-615-0914, bssid=(null), security=WPA2 Personal, rssi=-84, channel=<CWChannel: 0x600002095390> [channelNumber=13(2GHz), channelWidth={40MHz(-1)}], ibss=0]
)}
```

I do not have any 14.3 systems at hand but working on virtualizing Sonoma.
Would you by chance have a MacOS CI/CD test pipeline or similar to compare versions ?

tcamise-gpsw, 2024-04-10:
I also am getting null SSID's on Sonoma 14.4.1. 
Is my understanding correct that this is because location services are not enabled? At least on my machine, I do not appear to have anyway of enabling them.

epheo, 2024-04-10:
At least on the macbook someone landed me `location_manager.startUpdatingLocation()` is opening a graphical pop-up where you could approve the request for location services permission.
I also tested this an another 14.4.1 macintosh and got similar results with correct SSIDs.

However this inconsistency seems to corroborate others people findings as mentioned in:
> * [CoreWLAN returning None for SSID/BSSID ronaldoussoren/pyobjc#600](https://github.com/ronaldoussoren/pyobjc/issues/600)
> * https://forums.developer.apple.com/forums/thread/748518

I am not familiar with Apple's operating system and hardware but if, indeed, their low level tooling is unable to provide consistent results another option could be to adopt the same approach as for Windows and just try to connect until it either succeed or the maximum amount of attempts is reached.

tcamise-gpsw, 2024-04-10:
I do think the finite retries combined with your[ check for location authorization](https://github.com/epheo/OpenGoPro/commit/eeeb1328f66b10dc6e65a2f18cb018e0598a35fd) is the only path forward right now.
Regardless I still need to find a way to enable location services on my Mac.

keithrbennett, 2024-04-18:
I am very interested in a solution to this too. It seems weird to me that special permissions are now required to access information that the logged in user can already access using the GUI.  Also weird that Apple would choose to remove functionality from one utility without providing it in another one. I'm the author of a Ruby command line utility that reports and manages wifi state (https://github.com/keithrbennett/wifiwand) and this is a huge problem for my users and me too.

In addition to removing the ability to display all available networks, the removal of the `airport` utility also eliminates the abilities to do the following, which were, until now, supported in `wifiwand`:

* disconnect from the currently connected wifi network
* get additional information that `airport -I` provided. (I can't include which information that is because there is no longer a way I can get it.)

epheo, 2024-04-20:
cf this PR that may be worth testing
https://github.com/keithrbennett/wifiwand/pull/25

keithrbennett, 2024-04-20:
FYI, I have been able to restore wifi functionality by using Swift scripts, shelling out to them and then ingesting and parsing their input. However, these Swift scripts use CoreWLAN and, I believe, will not work unless XCode is installed. Here is the script that lists the names of available networks:

```swift
import Foundation
import CoreWLAN

class NetworkScanner {
    var currentInterface: CWInterface
    
    init?() {
        // Initialize with the default Wi-Fi interface
        guard let defaultInterface = CWWiFiClient.shared().interface(),
              defaultInterface.interfaceName != nil else {
            return nil
        }
        self.currentInterface = defaultInterface
        self.scanForNetworks()
    }
    
    func scanForNetworks() {
        do {
            let networks = try currentInterface.scanForNetworks(withName: nil)
            for network in networks {
                print("\(network.ssid ?? "Unknown")")
            }
        } catch let error as NSError {
            print("Error: \(error.localizedDescription)")
        }
    }
}

NetworkScanner()
```

keithrbennett, 2024-04-25:
@sfoley-gpqa You may just need to install XCode's command line tools: `xcode-select --install`. See https://www.freecodecamp.org/news/install-xcode-command-line-tools/ for an article about it.

sfoley-gpqa, 2024-04-25:
@keithrbennett 

My team has had that installed for years 😬

```shell
xcode-select --install
xcode-select: note: Command line tools are already installed. Use "Software Update" in System Settings or the softwareupdate command line interface to install updates
```

keithrbennett, 2024-04-25:
@sfoley-gpqa:

**Issue 1**

What version do you get when you run `pkgutil --pkg-info=com.apple.pkg.CLTools_Executables`? I get 15.3.0.0.1.1708646388 and my system says I am up to date.

When you run `swift --version`, what do you get? I get:
`Apple Swift version 5.10 (swiftlang-5.10.0.13 clang-1500.3.9.4)`

I did some brief research and this error occurred in prior versions of Swift. I'm hoping that your version is not up to date, and an update will fix it?

**Issue 2**

When running `swift` by itself, I get some help output. What is it you pasted the code into? `swift repl` starts a text mode REPL, is that what it was?  Or did you mean XCode?

**Issues 3 & 4**

I had the same result. Github Copilot had the information and advice pasted below. This is more trouble than it's worth for me at the moment, since for my purposes it's ok not to compile the script:

----

The issue might be related to the permissions required to access WiFi information on macOS. When you run a Swift script using the Swift interpreter (i.e., `swift script.swift`), it runs in a sandboxed environment which has limited access to system resources. This is a security feature of macOS.

However, when you compile a Swift program into a binary and run it, it's not sandboxed in the same way. Therefore, it might not have the necessary permissions to access the WiFi information.

To resolve this issue, you can try to add the necessary entitlements to your Swift program and then codesign it. This process involves creating an entitlements file, compiling your Swift program into a binary, and then signing that binary with the entitlements file.

Here's a step-by-step guide:

1. Create an entitlements file named `entitlements.plist` with the following content:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>com.apple.wifi.scan</key>
    <true/>
</dict>
</plist>
```

2. Compile your Swift program into a binary:

```bash
swiftc -o networkscanner AvailableWifiNetworkLister.swift
```

3. Sign the binary with the entitlements file:

```bash
codesign --entitlements entitlements.plist -s - networkscanner
```

Please note that this process requires you to have a valid Developer ID Application certificate in your keychain. If you don't have one, you can create one in the Apple Developer portal. Also, keep in mind that this process might not work if your app is distributed outside of the Mac App Store, as the necessary entitlements might not be granted.

----

tcamise-gpsw, 2024-04-25:
@keithrbennett Here's a strange half-baked idea. Since you're already managing this in the WifiWand package, I wonder if it would be feasible for me to consume WifiWand in my Python SDK here. I guess this would be something like:
1. Compile WifiWand to binary
2. Ship binary as part of Python SDK when installing on MacOS

Are you aware of any obvious pitfalls to this idea? Specifically:
1. Will I need to compile per MacOS version or do any other special handling for MacOS versions?
2. Will the compiled WifiWand handle gathering the necessary permissions?

keithrbennett, 2024-04-26:
@sfoley-gpqa 

### Re: Issue 1

It seems that those tools were installed in Februrary, does that sound right? This is your installation timestamp converted to date in Ruby's `irb` (REPL):

```ruby
:001 > Time.at(1707601182).utc
 => 2024-02-10 21:39:42 UTC
```

Is it possible that you `brew install`ed swift and that is what is being run? Does `brew install swift` show that it is (brew) installed?

### Re: Issue 2

I'm surprised that you get that prompt when you run `swift` by itself. I get:

```
$ swift

Welcome to Swift!

Subcommands:

  swift build      Build Swift packages
  swift package    Create and work on packages
  swift run        Run a program from a package
  swift test       Run package tests
  swift repl       Experiment with Swift code interactively

  Use `swift --version` for Swift version information.

  Use `swift --help` for descriptions of available options and flags.

  Use `swift help <subcommand>` for more information about a subcommand.
```

### Re: Issues 3 & 4

I used Swift because it was simpler to run the Swift interpreter on a (text) source code file than to build a Mac executable with all the ceremony that entails. Your use case might justify using Objective C instead; it is lower level than Swift and may require fewer setup steps for your users. Also, for either language, if you built an application (rather than interpreting a script), then you would only need all the setup on a single dev machine to build the app. I realize that Swift is not working for built applications -- it's possible that Objective C would work ok, I don't know.

### Re: Copilot

I tried building and running the application, and for me too, when I ran it, it immediately exited. I suspect that the OS knew that it was not a legit app and killed it. Fixing it may be as simple as getting a developer registration. I'm not thrilled about having to do that either. ;)

### Re: Calling WifiWand from Python

Although you could certainly call a Ruby (or any other) executable from Python, there is no way that I know of to compile a Ruby application to binary. Also, you would still need to install all the Ruby infrastucture and gems (wifiwand plus its dependencies).

It wouldn't help with building Ruby into a binary, but in other ways, it's _possible_ that JRuby would work better for you, I don't know. JRuby is Ruby that is run by a Ruby interpreter written in Java and running on the JVM. From the point of view of the system, the Ruby app is really a Java app, so the configuration issues might be easier to resolve, since Java use is widespread. If your users' machines already have Java installed, and/or your organization uses Java applications, this might be worth looking into. However, it takes a couple of seconds for a JVM to start up, and that may be a dealbreaker for you.

My gut feeling is that your best bet is to write something in Python yourselves. The wifi reporting and management part of wifiwand (as opposed to its UI) is quite simple and consists mostly of running Mac OS external applications. If you extract that functionality and tailor it for your specific use case, then I believe it would not be complex or overly time consuming. Regarding the Swift part, it's possible that Objective-C might turn out to be a better option if, for example, the CoreWLAN functionality can be statically linked into the distributed application, eliminating the need for it to be found on the user's system at runtime.

I am available to help on a consulting basis if that would be helpful.

VladislavGatsenko, 2024-07-25:
I was looking for a solution to a different problem, but maybe this will help with a solution. Just execute in the console:

system_profiler SPAirPortDataType

keithrbennett, 2024-07-25:
Vladislav -

Very interesting! And using system_profiler's `-json` option, the output is machine readable.

Using my `rexe` gem (https://github.com/keithrbennett/rexe), which does format conversions, can pull in the JSON as a hash, and can execute Ruby commands on a single command line , one can output just the network information, in pretty JSON mode, with the command:

system_profiler -json SPAirPortDataType | rexe -ij -mb -oJ 'self["SPAirPortDataType"].first["spairport_airport_interfaces"].first["spairport_airport_other_local_wireless_networks"]'

[
{
    "_name": "dlink-204C-5GHz",
    "spairport_network_channel": "149 (5GHz, 80MHz)",
    "spairport_network_phymode": "802.11",
    "spairport_network_type": "spairport_network_type_station",
    "spairport_security_mode": "spairport_security_mode_wpa2_personal_mixed",
    "spairport_signal_noise": "-93 dBm / -96 dBm"
  },
...
]

(Although this works, it assumes a specific order of the hash content, so a more robust implementation is really needed; but this illustrates the concept.)

- Keith

sfoley-gpqa, 2024-07-25:
> I was looking for a solution to a different problem, but maybe this will help with a solution. Just execute in the console:
> 
> system_profiler SPAirPortDataType

That's interesting; I just found the same thing last week and recently integrated it into an internal SDK. I was going to post about it here presently but it seems you beat me to the punch!

And Keith: Hahaha, I spent several hours trying to implement dual `re` and `pyparsing` solutions to parse out the information from the output of `system_profiler SDAirPortDataType`. I had no idea there was a JSON output flag. You're the best 🎉

sfoley-gpqa, 2024-07-25:
If it's useful, here is my `re` solution. I couldn't get the `pyparsing` one to work with SSIDs that have colons in the name (e.g. `"SSID:with:colon"`) after a few tries and gave up since `re` worked.

```python
def scan_for_ssid(ssid: str, timeout_sec: float = 30.0, invert: bool = False) -> bool:
    """
    Determine if the local host can find a specific SSID by name

    :param ssid: The name of the SSID to scan for
    :param timeout_sec: How long to scan (seconds) before giving up (plus duration of one scan, which can take few sec)
    :param invert: If True, verify that the SSID is _not_ found within the timeout
    :return: True if the SSID was found (unless invert==True); False otherwise
    """

    if timeout_sec is None:
        timeout_sec = 10.0

    path_system_profiler = Path('/usr/sbin/system_profiler')
    current_platform = platform.system()

    if current_platform != 'Darwin':
        raise EnvironmentError(f'Platform not supported (yet): {current_platform}')
    if not path_system_profiler.exists():
        raise EnvironmentError(f'{path_system_profiler} not found. Unable to scan for SSIDs')

    ssids: list[str] = []

    print(f'Scanning for SSID: "{ssid}" (timeout: {timeout_sec:.2f} seconds)')
    start_time = time.time()
    while time.time() - start_time <= timeout_sec:
        output: bytes = subprocess.check_output([str(path_system_profiler), 'SPAirPortDataType'])
        text: str = output.decode()
        regex = re.compile(r'\n\s+([\x20-\x7E]{1,32}):\n\s+PHY Mode:') # 0x20...0x7E --> ASCII for printable characters
        for _ssid in sorted(list(set(regex.findall(text)))):
            if _ssid not in ssids:
                ssids.append(_ssid)
            if not invert and ssid in ssids:
                return True
        #print(f'Scanning... ssids found: {sorted(ssids)}')  # debug

    return invert
```

tcamise-gpsw, 2024-07-25:
This is good to hear. I'm currently doing some general SDK maintenance and regression testing on MacOS so will bring this in.

autopulated, 2024-08-22:
@sfoley-gpqa thank you, this works for me on 14.6.1 👍

tcamise-gpsw, 2024-09-09:
I've merged this into the Python SDK. I'm fairly close to extracting the Wifi "Driver" out into its own package at which point the Python SDK will import it. But this won't happen for a few weeks at least.

closed 2024-09-09 by tcamise-gpsw