gopro/OpenGoPro #509: Add conditions to Github Actions to avoid running workflows on forks

epheo, 2024-04-10:
As per the example from GitHub docs:
https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#example-only-run-job-for-specific-repository

It would be better to add a conditions in the Github actions workflows in order for them to run only on `gopro/OpenGoPro` and prevent them to run on forked repositories.

```
if: github.repository == 'gopro/OpenGoPro'
```

tcamise-gpsw, 2024-04-10:
I think I've found the root of the problem. Currently the copyright action runs on each PR update and updates the PR branch with missing copyrights. There is some [strange GH Action behavior](https://github.com/orgs/community/discussions/25702) where actions will not get retriggered using the default GH token. Therefore I have been using a repository secret token. However this secret token is of course not exposed to forks. 

I think the solution is to move the copyright action out of the PR check and into the "push to main" check. Of course all of the runners seem to be down right now so I can't test yet...

tcamise-gpsw, 2024-04-10:
I just tested on a fork I made and it seems to be working. I see no reason why it would fail for you but you should try it anyway

closed 2024-04-10 by tcamise-gpsw