gopro/OpenGoPro #519: GoPro Wifi sometimes stops responding during long sessions

pavlosharhan2, 2024-04-12:
There is a bit mysterious issue. The WiFi http API stops working and starts responding with {'Unknown error.'}. Once it happens there are 3 possible ways to recover:

Just wait like 15 minutes and It recovers by Itself (helps in 20% of cases)
Restart the camera ( Power off/ Power on ) -  (helps in 80% of cases)
Reset connections on the camera - always helps

This doesn't happens all the time. But when it happens, It' usually after 3-4 hours of continuous work.
This is the only one problem that blocks us from controlling the camera in fully unattended mode

It would be great to have at least some understanding of what may cause this {'Unknown error.'}. I am keen to investigate further to give more input

pavlosharhan2, 2024-04-22:
Further investigation suggests that it starts happening when  there are some troubles with sd card.
When I see. "Unknown Error", the number of clips on camera in the status (code 39) shows a bigger number than It is in reality. I see something like 70 clips on camera, while the sd card is empty. But the SD card status ( code 33) shows OK.
So i  think the only problem here is that sd_card_status ( code 33 ) doesn't always indicate that there is some problem with the storage.

Formatting sd card fixes the problem

tcamise-gpsw, 2024-04-23:
Interesting, thanks for the investigation. Certainly seems like a valid bug so I will mark it as such.

pavlosharhan2, 2024-04-23:
@tcamise-gpsw Have added another type of sympthom besides the "Unknown error"

pavlosharhan2, 2024-06-21:
I've finally understood the problem.

**Preconditions:**
**Camera model:** HERO10 
So It occurs when you power the camera on via USB cable while having a battery inside with less than 20% level OR just when using GoPro Labs firmware with "Trust USB" QR code so the camera can be powered solely from USB.

**Steps to reproduce:**
Just connect USB cable to the camera, wait about 15 sec until the camera goes in "hibernation state" and just waits for BLE connection, then Connect to It via BLE ( No matter if IT's Quick App or some custom BLE client) - DONE. Now when you try to connect via wifi and call http://10.5.5.9/gopro/media/list - you get an "Unknown error." response. When you try to view the media in the app - yo'll get "Low battery" pop up and you won't be able to  view the media.

**Workaround:**
The fix is to use "Wake on Power" ("WAKE") Labs QR code. This way the camera is always powered on automatically when there is some power supplied, and you can successfully query the media list.

**Question:**
Why don't the camera works normally without this "Wake on power" feature enabled? Is there any reason the media list can't be queried if the power was turned on by BLE connection and not by Wake on power/pressing the side button?

pavlosharhan2, 2024-06-21:
Here’s a corrected version with improved grammar:

---

I'm still interested in gaining more knowledge about why the camera starts responding with "Unknown Error" during long sessions and how to prevent this from happening. Using the camera in a remote location seems a bit complicated, but it appears to be possible.

To understand how to prevent this, I would like to know the following:
1) Is there significant harm in pulling the media list every second? Do I need to have a minimal interval between requests, similar to the BLE API?
2) Could there be harm in requesting the media list via the HTTP API or removing files while the camera (HERO 10) is recording?
3) Are there any other measures I should take to protect against this?

---