gopro/OpenGoPro #521: OpenGoPro command response documentation incomplete

aaramirez-g, 2024-04-16:
What is the bug in?
- Documentation

**Describe the bug**
The new OpenGoPro BLE documentation is missing the Command Response Error codes that were present in the previous version. For example the Set [AP Control](https://gopro.github.io/OpenGoPro/ble/features/control.html#set-ap-control) response link just points to (https://gopro.github.io/OpenGoPro/ble/protocol/ble_setup.html#ble-characteristics) which does not give any details on what the response is. 

Previous Version: https://github.com/gopro/OpenGoPro/blob/90e8965f36f0233f17768e0ca04f9a17e5b5136a/docs/specs/ble_versions/ble_2_0.md#command-response

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

tcamise-gpsw, 2024-04-16:
good catch. I'll add these back in

tcamise-gpsw, 2024-04-16:
I've added the table of Command Statuses to the [Command TLV documentation](https://gopro.github.io/OpenGoPro/ble/protocol/data_protocol.html#commands).

Since these are common to all TLV Commands, they are not documented in each command. The idea is that you click on the TLV Type, i.e. TLV Command for "Set AP Control":

[image]

to then see this common information.

closed 2024-04-16 by tcamise-gpsw