gopro/OpenGoPro #523: Pairing mode after restart

ruzrulit, 2024-04-17:
I have a mobile app that set a livestream on gopro via BLE. 

**Case**: when the battery goes low, i need to replace it. After replacement the stream should continue automatically without any interaction. 

**Problem**: the connection could not be established automatically because the pairing mode is off after battery changing and a ble manager cannot find it during the scanning.

**Question**: is it possible to set gopro camera that way, so the pairing mode will be on all the time? or is there any other solution? or technically now it is not possible?

gstewart-gpsw, 2024-04-17:
If you have paired once, you should be able to connect after battery replace. The pairing keys are stored and not dependent on the battery.

tcamise-gpsw, 2024-07-01:
Can someone provide a list of steps to replicate the issue here following the [Bug Report Template](https://github.com/gopro/OpenGoPro/issues/new?assignees=&labels=bug&projects=&template=bug_report.md&title=Bug+Description)?

What I can gather from the conversation above is that the GoPro can not be re-paired to via BLE after pulling and reinserting the battery without some type of UI interaction? This is certainly not true so I must be missing a step somewhere.

closed 2024-08-11 by github-actions[bot]