gopro/OpenGoPro #528: Hero 8 - Error while opening

alduxvm, 2024-04-22:
What is the bug in?
- Demo: `gopro-photo`
- Specs (BLE , cable and wifi)
- OS: Ubuntu 22

Trying to run the demo of `gopro-photo` I get the ` Error while opening` and the image never gets saved. 

Steps to reproduce the behavior:
1. Go to a terminal and type: 'gopro-photo'
2. Camera is found
3. Camera connects
4. See error

Log:
```
gopro-photo 
Need to run as sudo. Enter password: 
INFO     16:20:58.113 Scanning for GoPro [A-Z0-9]{4} bluetooth devices...                                                                                               bleak_wrapper.py:120
INFO     16:20:58.237    Discovered: 67:AF:BC:69:36:7B: 67-AF-BC-69-36-7B                                                                                               bleak_wrapper.py:133
INFO     16:20:58.240    Discovered: CE:DD:1F:56:52:11: GoPro 0947                                                                                                      bleak_wrapper.py:133
INFO     16:20:58.243    Discovered: 4E:22:F8:44:C1:04: 4E-22-F8-44-C1-04                                                                                               bleak_wrapper.py:133
INFO     16:20:58.247 Found 1 matching devices.                                                                                                                         bleak_wrapper.py:146
INFO     16:20:58.250 Establishing the BLE connection                                                                                                                          client.py:106
INFO     16:20:58.252 Establishing BLE connection to CE:DD:1F:56:52:11: GoPro 0947...                                                                                   bleak_wrapper.py:213
INFO     16:21:02.371 Pairing with bluetoothctl                                                                                                                         bleak_wrapper.py:262
INFO     16:21:02.648 Discovering characteristics...                                                                                                                    bleak_wrapper.py:358
INFO     16:21:03.532 Done discovering characteristics!                                                                                                                 bleak_wrapper.py:403
INFO     16:21:03.535 Enabling notifications...                                                                                                                         bleak_wrapper.py:321
INFO     16:21:04.340 Done enabling notifications                                                                                                                       bleak_wrapper.py:327
INFO     16:21:04.343                                                                                                                                                  gopro_wireless.py:716
         <<<<<<<<--------------                                                                                                                                                             
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                                                                                                                                     
             "status_id" : "StatusId.ENCODING",                                                                                                                                             
             "protocol" : "Protocol.BLE",                                                                                                                                                   
             "uuid" : "Query",                                                                                                                                                              
         <<<<<<<<--------------                                                                                                                                                             
                                                                                                                                                                                            
INFO     16:21:04.433                                                                                                                                                  gopro_wireless.py:649
         -------------->>>>>>>>                                                                                                                                                             
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                                                                                                                                     
             "status" : "ErrorCode.SUCCESS",                                                                                                                                                
             "protocol" : "Protocol.BLE",                                                                                                                                                   
             "data" : {                                                                                                                                                                     
                 "StatusId.ENCODING" : "False",                                                                                                                                             
             },                                                                                                                                                                             
         -------------->>>>>>>>                                                                                                                                                             
                                                                                                                                                                                            
INFO     16:21:04.438                                                                                                                                                  gopro_wireless.py:716
         <<<<<<<<--------------                                                                                                                                                             
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                                                                                                                                     
             "status_id" : "StatusId.SYSTEM_BUSY",                                                                                                                                          
             "protocol" : "Protocol.BLE",                                                                                                                                                   
             "uuid" : "Query",                                                                                                                                                              
         <<<<<<<<--------------                                                                                                                                                             
                                                                                                                                                                                            
INFO     16:21:04.520                                                                                                                                                  gopro_wireless.py:649
         -------------->>>>>>>>                                                                                                                                                             
             "id" : "QueryCmdId.REG_STATUS_VAL_UPDATE",                                                                                                                                     
             "status" : "ErrorCode.SUCCESS",                                                                                                                                                
[... cut ...]
INFO     16:21:09.711 Disconnecting...                                                                                                                                  bleak_wrapper.py:413
INFO     16:21:12.211 Device disconnected!                                                                                                                              bleak_wrapper.py:416
ERROR    16:21:12.213 TimeoutError() 
```

[image]

With wifi gets the same error (computer connected to the wifi of the computer and running using the --wifi_interface parameter)

With usb cable it just hangs...

**Hardware**
- Camera: Hero8 black

tcamise-gpsw, 2024-04-22:
It's because you're using an [unsupported camera](https://gopro.github.io/OpenGoPro/#compatibility)

closed 2024-04-22 by tcamise-gpsw

alduxvm, 2024-04-22:
Wow! dunno how I missed that...

Maybe that list could be added to the pip documentation? https://gopro.github.io/OpenGoPro/python_sdk/index.html

tcamise-gpsw, 2024-04-22:
no worries! 

Yeah that's probably a good idea. I'll add it. I also just updated the issue templates here since this happens pretty frequently