gopro/OpenGoPro #529: Hero 12 black STA mode error

NekoDeveloper, 2024-04-23:
[image]

I purchased a Hero12 Black camera and attempted to follow the example in the demo (demos/python/tutorial/tutorial_modules/tutorial_6_connect_wifi/connect_as_sta.py) to switch the camera to STA mode and connect to another AP hotspot. However, I encountered the error shown in the image while running the demo program. How did this error occur, and what aspects should I investigate to troubleshoot the issue?

NekoDeveloper, 2024-04-23:
I have found the reason for the issue. When in GoPro STA mode and connecting to another AP hotspot, it requires this hotspot to be connected to a public network, rather than being an AP hotspot within a local area network. OMG!!! I hope that such restrictions can be removed in the new firmware version.

tcamise-gpsw, 2024-04-23:
Hello. Yes this is a known issue: https://github.com/gopro/OpenGoPro/issues/467

I am working on a potential workaround here and will update that linked ticket when / if it works

closed 2024-04-23 by tcamise-gpsw

Ddezibel, 2024-11-29:
Hey, has there been any progress on this issue? I don't understand why this is linked to #467, since that is about accessing the camera from outside of the network with an internet connection, while this issue here is about accessing the camera from within the network without an internect connection.
I know my way around python, so if you point me into the right direction, I can look into the code myself. My gut feeling says it has to do with the DNS's in [proto/network_management_pb2.pyi, line 247](https://github.com/gopro/OpenGoPro/blob/c8b3664ced4951a3ccf33a7c1b8aaaff1a3ab128/demos/python/sdk_wireless_camera_control/open_gopro/proto/network_management_pb2.pyi#L247), but I'm not huge into networks unfortunately.

aoyurs, 2024-12-05:
I am encountering the same issue. I need to connect to a local network. Are there plans to support this in the future?

vuillod, 2026-09-09:
Hi everyone, 
Still facing the same issue to put the GoPro into COHN mode in a LAN setup (provisioning_state <EnumProvisioning.PROVISIONING_ERROR_FAILED_TO_ASSOCIATE: 7>). 
Any news ?