gopro/OpenGoPro #535: Wrong argument name for Delete Single Media File

romain-riviere, 2024-04-30:
**Component**
What is the bug in?
- **Documentation**

**Describe the bug**
I am requesting to Firmware H22.01.02.30.70 from a Flutter Application on Android 12

Here in the doc : https://gopro.github.io/OpenGoPro/http#tag/Media/operation/OGP_DELETE_SINGLE_FILE
Documentation specifies that we should use parameter "p" to specify the file to delete but the gopro response is HTTP 400.
I've tried with the parameter "path" and the gopro response is HTTP 200 and it actually deletes the file.

**Screenshots**
Responses in my logs :

Error:
[image]

Success:
[image]

**Hardware**
- Camera: Hero11
- Desktop: MacOS Sonoma
- Smartphone: Samsung S10

tcamise-gpsw, 2024-05-01:
Good catch. Let me verify and then I'll update the docs

tcamise-gpsw, 2024-05-09:
This has been fixed.

closed 2024-05-09 by tcamise-gpsw