gopro/OpenGoPro #536: Unable to

ajchandler39, 2024-04-30:
My ultimate goal is to have my gopro act as a webcam with a wired connection to my android, but I'm testing it with my PC first. I'm plugging my GoPro 10 into my laptop via USB connection and both trying to just do a ping to 172.29.155.51 (serial # ends in 955) and sending a request via postman to http://172.29.155.51:8080/gopro/camera/control/wired_usb?p=1. Neither are working. My GoPro 10 is being recognized on my PC; I can access videos, although my GoPro isn't saying it's connected via USB like it was on my GoPro 8.

Any help would be greatly appriciated.

tcamise-gpsw, 2024-05-01:
Seems to be working for you so I'll close.

FYI there is a python demo to do exactly what you are asking for:
https://github.com/gopro/OpenGoPro/tree/main/demos/python/multi_webcam

closed 2024-05-01 by tcamise-gpsw

pavloshargan, 2024-10-04:
@ajchandler39  your android device is likely not supporting CDC NCM. Try a different phone. You can my Kotlin app that handles USB connection automatically: https://github.com/pavloshargan/GoPro-Stream-USB-Android