gopro/OpenGoPro #537: PROVISIONING_ERROR_FAILED_TO_ASSOCIATE in connect_as_sta.py

onigiri5884, 2024-05-02:
I tried to execute connect_as_sta.py in [tutorial 6](https://gopro.github.io/OpenGoPro/tutorials/connect-wifi#station-sta-mode), but RuntimeError('Unexpected provisioning state: 7') was occured many times.
The provising state 7 is PROVISIONING_ERROR_FAILED_TO_ASSOCIATE.

Can you think of anything that might have caused it?

tcamise-gpsw, 2024-05-08:
There was a bug at one point in Hero 12 that would return this error but it was fixed.
Can you make sure you're on the latest firmware version?

If that doesn't fix it, then please post your entire log here.

onigiri5884, 2024-05-13:
Sorry for the late reply.

I'm using the latest firmwarevirsion v2.20 but Wi-Fi router is not connect to the internet.
(For security reasons, I want to control multiple GoPro on local network.)  

It seems that this problem is as same as issue [#529](https://github.com/gopro/OpenGoPro/issues/529).

Please tell me why it requires that the Wi-Fi router to be connected to the internet when connecting GoPro to a Wi-Fi router as STA mode?

closed 2024-06-22 by onigiri5884