gopro/OpenGoPro #538: Incorrect Documentation for Status Response 35

sepp89117, 2024-05-03:
According to your documentation, Status Response 35 returns the available storage space on the SD card in video minutes. However, in practice, it returns the remaining video time in seconds instead of minutes. This inconsistency can lead to confusion for developers integrating with the OpenGoPro API.

Documentation Link: [OpenGoPro Documentation](https://gopro.github.io/OpenGoPro/ble/features/statuses.html#how-many-minutes-of-video-can-be-captured-with-current-settings-before-sdcard-is-full-35)

Tested with Hero 10 and 12

tcamise-gpsw, 2024-05-03:
Good catch. I'll update it. Really all of the statuses need a review.

sepp89117, 2024-05-03:
Hi, since the revision of the documentation I also miss the data types.

tcamise-gpsw, 2024-05-09:
I've update minutes to seconds where applicable.

It is a good point that we should included data types (i.e. int16, etc).
I'll leave this ticket open as an enhancement request to add these.

tcamise-gpsw, 2024-07-10:
Integer data types have been added for [settings](https://gopro.github.io/OpenGoPro/ble/features/settings.html) and [statuses](https://gopro.github.io/OpenGoPro/ble/features/statuses.html#status-ids)

closed 2024-07-10 by tcamise-gpsw