gopro/OpenGoPro #547: Unable to change Access Point SSID

romain-riviere, 2024-05-15:
Hello,

I am building a Flutter application using FlutterBluePlus to communicate with Hero11 cameras.

I am trying to rename the GoPro SSID using the "WiFi AP SSID" characteristic from the "GoPro Wifi Access Point" service
[image]

This characteristic seems to be writable so i'm writing the new name to it. 

From these logs it seems to work : 
[image]

But, when I read again this characteristic, it's back to the old value :(

I've also tried to write on it using "BLE Scanner" from Play Store but I've got the same result.

Is this writable characteristic actually editing the SSID ?
Is there an additional step to be able to edit the camera's SSID ? Or another way ?
I've been able to change it from the Quik app so I know it's possible from BLE.

Regards,
Romain

romain-riviere, 2024-05-16:
By the way, I have tried this on two different Hero11 with two different firmware : 
- H22.01.02.10.00 (default firmware)
- H22.01.02.30.70 (labs firmware)

Both have OpenGoPro version 2.0

tcamise-gpsw, 2024-07-01:
Hi. This is not currently supported in Open GoPro but I will leave this open as an Enhancement Request

mlaity, 2024-10-21:
It would also be helpful to know how to reset a camera's AP back to the original SSID.  We have users renaming their cameras all the time, but our apps primarily rely on the typical Serial number based SSID.  Would be nice to just reset to original.

Thanks!