gopro/OpenGoPro #550: State status key 111 reports slow SD card for fast SD card (Hero 12 Black)

tony-gutierrez, 2024-05-17:
> First ensure you are using a camera that is [currently supported](https://gopro.github.io/OpenGoPro/#compatibility) by Open GoPro. Any other tickets will be rejected.

**Component**
What is the bug in?
- Specs (BLE, HTTP)

**Describe the bug**
I am using a Hero 12 Black with the latest GP labs fw.  The camera state is reporting a slow SD card. This is the card I am using:

[image]

**To Reproduce**
Steps to reproduce the behavior:
Install this card, and fetch camera state. Look at status[111].

**Expected behavior**
Is this card not suitable? Why is 111 reporting 1? Is the documentation for 111 wrong here? 
https://gopro.github.io/OpenGoPro/http#tag/Query/operation/OGP_GET_STATE
[image]

bfreeland-gpsw, 2024-05-29:
Hi @tony-gutierrez,

Status 111 is intended as a error flag to indicate a problem with the SD card's write speed. A value of 1 indicates the SD card write speed is too slow while a value of 0 indicates the write speed is acceptable. We will update the documentation in future to clarify this.

tcamise-gpsw, 2024-07-01:
I'll update the status description to "Is there a minimum SD Card write speed error?"

tcamise-gpsw, 2024-07-01:
This has been updated for [BLE ](https://gopro.github.io/OpenGoPro/ble/features/statuses.html#sd-card-write-speed-error-111) and [HTTP](https://gopro.github.io/OpenGoPro/http#schema/State)

closed 2024-07-01 by tcamise-gpsw