gopro/OpenGoPro #555: Calling `http_command.set_shutter(shutter=ENABLE)` while in COHN mode hangs indefinitely

sjdemartini, 2024-05-30:
### **Component**

What is the bug in?
- **Specs (BLE, HTTP)** / Demo

### **Describe the bug**

When using COHN mode following the [`cohn.py` demo](https://github.com/gopro/OpenGoPro/blob/28efd7c0d9e1bc43343bed5274cf74f84a2cded1/demos/python/sdk_wireless_camera_control/open_gopro/demos/cohn.py) and then calling `await gopro.http_command.set_shutter(shutter=Params.Toggle.ENABLE)`, the command hangs indefinitely and will never return. Using a keyboard interrupt shows it's stuck on this line https://github.com/gopro/OpenGoPro/blob/28efd7c0d9e1bc43343bed5274cf74f84a2cded1/demos/python/sdk_wireless_camera_control/open_gopro/gopro_wireless.py#L554:

```
---------------------------------------------------------------------------
CancelledError                            Traceback (most recent call last)
Cell In[8], line 1
----> 1 await gopro.http_command.set_shutter(shutter=Params.Toggle.ENABLE)

File /path/to/env/lib/python3.10/site-packages/open_gopro/api/builders.py:772, in http_get_json_command.<locals>.wrapper(wrapped, instance, _, kwargs)
    770 @wrapt.decorator
    771 async def wrapper(wrapped: Callable, instance: HttpMessages, _: Any, kwargs: Any) -> GoProResp:
--> 772     return await instance._communicator._get_json(message, rules=rules, **(await wrapped(**kwargs) or kwargs))

File /path/to/env/lib/python3.10/site-packages/open_gopro/gopro_wireless.py:765, in WirelessGoPro._get_json(self, message, *args, **kwargs)
    763 async def _get_json(self, message: HttpMessage, *args: Any, **kwargs: Any) -> GoProResp:
    764     message = self._handle_cohn(message)
--> 765     return await super()._get_json(*args, message=message, **kwargs)

File /path/to/env/lib/python3.10/site-packages/open_gopro/gopro_base.py:110, in enforce_message_rules(wrapped, instance, args, kwargs)
     97 @wrapt.decorator
     98 async def enforce_message_rules(wrapped: MessageMethodType, instance: GoProBase, args: Any, kwargs: Any) -> GoProResp:
     99     """Decorator proxy to call the GoProBase's _enforce_message_rules method.
    100 
    101     Args:
   (...)
    108         GoProResp: common response object
    109     """
--> 110     return await instance._enforce_message_rules(wrapped, *args, **kwargs)

File /path/to/env/lib/python3.10/site-packages/open_gopro/gopro_wireless.py:554, in WirelessGoPro._enforce_message_rules(self, wrapped, message, rules, **kwargs)
    552     if rules.should_wait_for_encoding_start(**kwargs):
    553         logger.trace("Waiting to receive encoding started.")  # type: ignore
--> 554         await self._encoding_started.wait()
    555         self._encoding_started.clear()
    556 return response

File /path/to/env/lib/python3.10/asyncio/locks.py:214, in Event.wait(self)
    212 self._waiters.append(fut)
    213 try:
--> 214     await fut
    215     return True
    216 finally:

CancelledError: 
```

### **To Reproduce**

Use the following Python code:

```python
import asyncio
from open_gopro import Params, WiredGoPro, WirelessGoPro, proto

# COHN setup from
# https://github.com/gopro/OpenGoPro/blob/28efd7c0d9e1bc43343bed5274cf74f84a2cded1/demos/python/sdk_wireless_camera_control/open_gopro/demos/cohn.py
wifi_ssid = ''
wifi_password = ''

# Start with Wifi Disabled (i.e. don't allow camera in AP mode).
async with WirelessGoPro(target='GoPro 1234', enable_wifi=False) as gopro:
    assert gopro
    await gopro.ble_command.cohn_clear_certificate()

    if await gopro.is_cohn_provisioned:
        print("COHN is already provisioned")
    else:
        assert await gopro.connect_to_access_point(wifi_ssid, wifi_password)
    assert await gopro.configure_cohn()
    print('Set up COHN! Disconnecting Bluetooth')

assert (await gopro.http_command.get_camera_state()).ok
print("Successfully communicated via COHN!!")

# Based on
# https://github.com/gopro/OpenGoPro/blob/28efd7c0d9e1bc43343bed5274cf74f84a2cded1/demos/python/sdk_wireless_camera_control/open_gopro/demos/video.py
assert (await gopro.http_command.load_preset_group(group=proto.EnumPresetGroup.PRESET_GROUP_ID_VIDEO)).ok

await gopro.http_command.set_shutter(shutter=Params.Toggle.ENABLE)  # <-- HANGS HERE

await asyncio.sleep(args.record_time) # <-- Never gets here
assert (await gopro.http_command.set_shutter(shutter=Params.Toggle.DISABLE)).ok
```

### **Expected behavior**

Like in BLE or regular WiFi AP mode, I'd expect `set_shutter(shutter=ENABLE)` to return once the shutter is on (not hang), so that I can then proceed with waiting a set duration before calling `set_shutter(shutter=DISABLE)` to stop the shutter and end the video.

I confirmed that I was able to `set_shutter(ENABLE)`, `asyncio.sleep`, and `set_shutter(DISABLE)` when using BLE commands, so this appears to be specific to COHN.

### **Hardware**
- Camera: HERO12 Black
- Python 3.10
- `open_gopro`: 0.16.1

sjdemartini, 2024-07-01:
@tcamise-gpsw Thanks for following up. Unfortunately I do not have the log files and am not working with the GoPro devices at this point, so can't easily run this again to produce a new set of them.

tcamise-gpsw, 2024-07-01:
No worries, actually I am able to replicate this easily. 

It's more or less the same problem as this [pinned issue](https://github.com/gopro/OpenGoPro/issues/487). Specifically the internal state management is not correctly reset / managed when the BLE connection drops after provisioning / connecting COHN. I'll do a little more investigating but will likely end up closing this as duplicate to be tracked with the pinned issue above which really needs to be dealt with soon.

tcamise-gpsw, 2024-07-01:
duplicate of #487

closed 2024-07-01 by tcamise-gpsw

MattL-NZ, 2025-02-13:
Hey @sjdemartini did you ever manage to get around this bug? I see that #487 is still open with no visible signs of it being fixed.

sjdemartini, 2025-02-13:
@MattL-NZ unfortunately not. I tried using bluetooth I believe, but soon after abandoned working with the GoPros altogether.