gopro/OpenGoPro #559: Bug Description

sleman, 2024-06-25:
**Component**
What is the bug in?
BLE Spec

**Describe the bug**
There is no way to set the GoPro to turn off Super Photo to capture a regular JPG?

**To Reproduce**
Steps to reproduce the behavior:
1. Activate BLE Photo Mode
2. Take photo
3. Photo will be captured as a Super Photo

**Expected behavior**
A way to change from Super Photo to regular Photo

**Hardware**
- Camera: All GoPro models
- Desktop if applicable [e.g. Windows]
- Smartphone if applicable [e.g. iPhone6]

**Additional context**
I also tried setting PHOTO_EASY_MODE (177) to 0 and I get an error 0x02.

tcamise-gpsw, 2024-07-01:
This setting is not currently exposed. I'll investigate documenting it and adding it to the Python SDK

tcamise-gpsw, 2024-07-01:
The setting is now documented via [BLE ](https://gopro.github.io/OpenGoPro/ble/features/settings.html#photo-output-125)and [HTTP](https://gopro.github.io/OpenGoPro/http#tag/settings/operation/GPCAMERA_CHANGE_SETTING::125) and added to the [Python SDK](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.api.ble_commands.BleSettings.photo_output).

closed 2024-07-01 by tcamise-gpsw