gopro/OpenGoPro #561: Add Protune settings

hankzoid, 2024-07-03:
I've built camera arrays using DSLRs and am looking to build one using 16-20 GoPros controlled via USB. I bought a Hero 12 to have a play with but have just realised it's not possible to control any of the protune settings - shutter speed, iso range, white balance etc. Obviously it's essential that all exposures match in a camera array so unfortunately I can't take this any further for now, unless I'm missing something?

I've seen that adding shutter speed was mentioned a couple of years ago. This seems like a no brainer so I'm sure there's a reason why it hasn't been added yet. Ideally I'd be able to set the white balance and shutter speed, and fix the ISO (or set the min and max range to the same value).

I'll keep an eye out for updates but if there are no plans to add this please let me know.

tcamise-gpsw, 2024-07-08:
Yeah unfortunately this is currently blocked due to non-technical reasons. I'm going to use this ticket as the main ticket to track any similar requests.

pavlosharhan2, 2024-07-12:
I have no isues in setting protune settings on GoPro 10/11/12 since year 2021.
It's even mentioned in the docs: https://gopro.github.io/OpenGoPro/ble/features/control.html#set-shutter

For example, the list of protune settings I'm able to set:

            "2" to "resolution",
            "3" to "fps",
            "145" to "shutter_speed",
            "102" to "iso_min",
            "13" to "iso_max",
            "115" to "white_balance",
            "116" to "color",
            "117" to "sharpness",
            "118" to "exposure",
            "121" to "lens",
            "124" to "bitrate",
            "88" to "lcd_brightness",
            
            And available options for each setting:
            
            val resolutionOptions = TwoWayDict(
            1 to "4K",
            4 to "2.7K",
            6 to "2.7K4:3",
            7 to "1440",
            9 to "1080p",
            18 to "4K4:3",
            24 to "5K",
            25 to "5K4:3",
            100 to "5.3K"
        )

        private val fpsOptions = TwoWayDict(
            0 to "240",
            1 to "120",
            2 to "100",
            5 to "60",
            6 to "50",
            8 to "30",
            9 to "25",
            10 to "24",
            13 to "200"
        )

        private val shutterOptions = TwoWayDict(
            0 to "Auto",
            3 to "1/24",
            4 to "1/25",
            5 to "1/30",
            6 to "1/48",
            7 to "1/50",
            8 to "1/60",
            11 to "1/96",
            12 to "1/100",
            13 to "1/120",
            16 to "1/192",
            17 to "1/200",
            18 to "1/240",
            25 to "1/384",
            21 to "1/400",
            22 to "1/480",
            28 to "1/800",
            23 to "1/960",
            29 to "1/1600",
            24 to "1/1920",
            30 to "1/3200",
            31 to "1/3840"
        )

        private val isoMinOptions = TwoWayDict(
            0 to "6400",
            3 to "3200",
            1 to "1600",
            4 to "800",
            2 to "400",
            7 to "200",
            8 to "100"
        )

        private val isoMaxOptions = TwoWayDict(
            0 to "6400",
            3 to "3200",
            1 to "1600",
            4 to "800",
            2 to "400",
            7 to "200",
            8 to "100"
        )

        private val whiteBalanceOptions = TwoWayDict(
            3 to "6500K",
            7 to "6000K",
            2 to "5500K",
            12 to "5000K",
            11 to "4500K",
            0 to "Auto",
            4 to "Native",
            5 to "4000K",
            10 to "3200K",
            9 to "2800K",
            8 to "2300K"
        )

        private val colorOptions = TwoWayDict(
            1 to "Flat",
            2 to "Natural",
            100 to "Vibrant"
        )

        private val lensOptions = TwoWayDict(
            7 to "Max SuperView",
            3 to "SuperView",
            0 to "Wide",
            4 to "Linear",
            8 to "Linear Horizon Leveling",
            2 to "Narrow"
        )

        private val sharpnessOptions = TwoWayDict(
            0 to "High",
            1 to "Medium",
            2 to "Low"
        )

        private val exposureOptions = TwoWayDict(
            8 to "-2.0",
            7 to "-1.5",
            6 to "-1.0",
            5 to "-0.5",
            4 to "0.0",
            3 to "0.5",
            2 to "1.0",
            1 to "1.5",
            0 to "2.0"
        )

        private val bitrateOptions = TwoWayDict(
            1 to "High",
            100 to "Standard"
        )

        private val lcdBrightnessOptions = TwoWayDict(
            10 to "10%",
            20 to "20%",
            30 to "30%",
            40 to "40%",
            50 to "50%",
            60 to "60%",
            70 to "70%",
            80 to "80%",
            90 to "90%",
            100 to "100%"
        )

hankzoid, 2024-07-17:
Thank you for this. I was really hoping I was wrong, but I can't see how to set these? The link that you provided refers to triggering the shutter to take a photo or start/stop recording, rather than changing the shutter speed.

I've tried using http requests and the values that you've posted, but I get a 404 error when trying to set the shutter speed. I get a 200 response when setting the auto power off time using the same process. Changing any of the Protune settings are also not mentioned here - https://gopro.github.io/OpenGoPro/http#tag/settings/operation/GPCAMERA_CHANGE_SETTING::59

For example -

```import requests
url = "http://172.{camera_url}:8080/gopro/camera/setting"
querystring = {"option":"4","setting":"59"}      # (Auto Power Down -> 5 mins)
response = requests.request("GET", url, params=querystring)
print(response)```

returns 200 and correctly changes the Auto Power Down value. However -

```import requests
url = "http://172.{camera_url}:8080/gopro/camera/setting"
querystring = {"option":"12","setting":"145"}      # (Shutter Speed -> 1/100)
response = requests.request("GET", url, params=querystring)
print(response)```

returns 404 error

If you could let me know how you're changing these settings I'd really appreciate it, ideally over USB/WiFi

hankzoid, 2024-07-17:
Ok interesting, I can set the shutter speed this way in video mode but not photo mode, I guess because there are different shutter options for video and photo e.g. 1/120 for video, 1/125 for photo. Where did you find this info for the shutter speed options?

pavlosharhan2, 2024-07-24:
> Ok interesting, I can set the shutter speed this way in video mode but not photo mode, I guess because there are different shutter options for video and photo e.g. 1/120 for video, 1/125 for photo. Where did you find this info for the shutter speed options?

The shutter speed should be a multiple of the FPS. You can't set 1/100 for 60 FPS. Your options are:
1/60
1/120
1/240
1/480
etc.

1/100 would be possible if you set anti-flicker to 50 Hz first (https://gopro.github.io/OpenGoPro/http#tag/settings/operation/GPCAMERA_CHANGE_SETTING::134), which would allow you to set FPS to 25/50/100. In this case, you could use a 1/100 shutter speed.

Another important thing: ensure Protune is on before setting the settings. For example, you can always call "gp/gpControl/setting/114/1" before applying any settings

BenSmithers, 2025-02-06:
> Ok interesting, I can set the shutter speed this way in video mode but not photo mode, I guess because there are different shutter options for video and photo e.g. 1/120 for video, 1/125 for photo. Where did you find this info for the shutter speed options?

I'm not sure if you're still working on this, @hankzoid , but I found a way to extract out the setting numbers in a procedural way on a Hero 10 (likely will be a very similar procedure!) while working on a project of my own. If you manually change the protune settings on your camera, then through Open GoPro save the camera state dictionary to disk 
```
state = await gopro.http_command.get_camera_state()
out_dict = {}
out_dict["id"] = str(state.identifier)
out_dict["status"] = str(state.status)
out_dict["protocol"] = str(state.protocol)
out_dict["data"] = {str(key):str(state.data[key]) for key in state.data}

import json 
_obj = open("./resources/state_800min.json", 'wt')
json.dump(out_dict, _obj, indent=4)
_obj.close()
```
you can then change the setting again and output a second state json. I did this in nightmode before and after changing the ISO min from 400 to 800, and ran a diff on the json files. 
```
124c124
<         "SettingId.INTERNAL_75": "0",
---
>         "SettingId.INTERNAL_75": "1",
```
After repeating for multiple values I found that a `SettingId.INTERNAL_75` of 3 gives 100 iso, 2 gives 200 iso, 1 gives 400, and 0 gives 800. I could then change those settings on the camera by running
```
curl --request GET --url 'http://172.29.134.51:8080/gopro/camera/setting?setting=75&option=0'
```
on my computer with the desired "option" for my ISO. The above set it to 800.

Likewise, I found that setting 24 tweaks the ISO Max, setting 90 toggles the auto for the shutter speed, setting 19 changes the shutter speed itself, and so on and so on. These might not be the same setting for different modes, but this seems like it would generalize.