gopro/OpenGoPro #563: `gopro-livestream` demo not working. set shutter enable raises `TimeoutError`

LordXerus, 2024-07-04:
> First ensure you are using a camera that is [currently supported](https://gopro.github.io/OpenGoPro/#compatibility) by Open GoPro. Any other tickets will be rejected

GoPro Hero 10 Black is currently supported
**Component**
What is the bug in?
**- Demos (python sdk, etc)**
- Tutorials
- Documentation
- Specs (BLE, HTTP)

**Describe the bug**
When running the demo, the GoPro camera returns some unknown responses after connecting to the Wi-Fi hotspot. Then it times out on the Shutter Enable line of the code. The GoPro camera also stop responding to touch shortly after.

We were able to livestream to the same server via the RTMP option on the Quik App.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a suitable RTMP server. My laptop is running Arch Linux with an nginx server + nginx rtmp mod. VLC is used to playback the stream.
2. We used a hotspot on an Android phone to connect the laptop and the gopro.
3. Run the `gopro-livestream` demo with ssid, wifipass, and rtmp url. The rtmp url we used is `rtmp://192.168.219.104/live`. This setting works on the Quik app, but not in the demo. We are running the demo off the same laptop.
4. Sometimes the demo cannot find the GoPro camera. In this case we swipe down > swipe left > Connections > Connect Device > GoPro Quik App.
5. After the GoPro connects to the hotspot, it errors out on setting up the live stream.
6. See error
Output of `gopro-livestream ... | tee gopro-livestream-out.txt`
[gopro-livestream-out.txt](https://github.com/user-attachments/files/16103186/gopro-livestream-out.txt)

[gopro_demo.log](https://github.com/user-attachments/files/16103189/gopro_demo.log)

**Expected behavior**
The live stream should start on our server, like how it does with the Quik app.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware**
- Camera: GoPro Hero 10 Black
- ~~Desktop~~ Laptop if applicable: Laptop with Arch Linux + Python 3.11 + Nginx
- Smartphone if applicable (for Quik App): iPhone 13
- Smartphone for hotspot: Google Pixel 7

Laptop runs Nginx + RTMP mod + VLC
Phone runs hotspot
Laptop w/ demo or iPhone w/ Quik app sends BLE to GoPro camera
GoPro camera streams back to Laptop via Android phone hotspot

**Additional context**
Add any other context about the problem here.

tcamise-gpsw, 2024-07-08:
Hi. I'm able to replicate and see what the problem is.
The Python SDK is hardcoding the `window_size` and `len` arguments of the [set_livestream_mode](https://gopro.github.io/OpenGoPro/python_sdk/api.html#open_gopro.api.ble_commands.BleCommands.set_livestream_mode) API if they are not set from the command line.
The demos are only tested against the latest camera which in this case can support `window_size` and `lens` parameters that are not supported on the Hero 10. And these nonsupported Hero 10 parameters are bricking the camera. This is a camera bug as it should handle bad parameters gracefully.
In any case, we can avoid this in the Python SDK by only sending valid parameters. For now I will just not send any parameter if not set from the CLI and move the responsibility to the user to choose correct parameters. I just tested this and the livestream demo is working. I'll push an update on the PR linked here.

closed 2024-07-08 by tcamise-gpsw

tcamise-gpsw, 2024-07-18:
just pushed 0.16.2 which has the updates