gopro/OpenGoPro #568: ModuleNotFoundError: No module named 'tutorial_modules'

George-Gi, 2024-07-13:
Probably I am missing something obvious, but I have been struggling for a while with this.
I can't execute some demos.

```
python tutorial_4_ble_queries/ble_query_poll_resolution_value.py
Traceback (most recent call last):
  File "C:\Users\OpenGoPro-2.0\demos\python\tutorial\tutorial_modules\tutorial_4_ble_queries\ble_query_poll_resolution_value.py", line 12, in <module>
    from tutorial_modules import GOPRO_BASE_UUID, connect_ble, Response
ModuleNotFoundError: No module named 'tutorial_modules'
```

If I try to execute the **tutorial_1_connect_ble/ble_connect.py** works, but this one doesn't refer to the tutorial_modules.

tcamise-gpsw, 2024-07-17:
This looks like you just haven't setup your environment correctly. Have you followed the [setup](https://gopro.github.io/OpenGoPro/tutorials/connect-ble#setup) steps?

George-Gi, 2024-07-24:
> This looks like you just haven't setup your environment correctly. Have you followed the [setup](https://gopro.github.io/OpenGoPro/tutorials/connect-ble#setup) steps?

Thank you for your answer. I tried again the instructions, but I used a virtual environment this time and it worked.

closed 2024-07-24 by George-Gi