gopro/OpenGoPro #574: Provide way to circumvent SOP

maledies, 2024-07-24:
**Is your feature request related to a problem or use case? Please describe.**
I use two Hero12. Since "The Remote" is not working in a multicam-context regarding the Hero12 and GoPro is not able to fix the issue for over 10 months (while still marketing it as "compatible" and dishonestly communicating to its userbase! just had to be said), I implemented a *browser-based multicam remote* using the Web Bluetooth API. It works quite well! So I wanted to extend the functionality using the HTTP API (more range, maybe directly downloading/showing taken images etc). But the same-origin policy (SOP) kicks in and API requests fail, because CORS is not implemented.

**Describe the solution you'd like**
The camera should support CORS. Since it is just about some headers and an adequate OPTIONS response for preflights computational ressources should not be problematic(?). It would allow to control the camera from within a browser.

**Describe alternatives you've considered**
Not using `fetch`, but opening the URL (like `/gopro/camera/shutter/start`) directly in a background tab/popup. Because multicam is a requirement, only COHN came into consideration, but COHN requires to set the `Authorization` header. It is not possible to provide the header when triggering a navigation (e.g. opening a tab) and URL-embedded credentials aren't support anymore. So the solution could be an endpoint like `GET /gopro/authenticate` which basically returns `WWW-Authenticate`. The user would enter the password once into the browser and the browser autmatically sends it along the following navigations. But overall it's rather an interim solution!

tcamise-gpsw, 2024-07-24:
I also agree this would be nice. The lack of CORS also prevents Swagger from being able to send test commands.
It's been specifically blocked for security reasons but perhaps these are outdated / not worth the tradeoffs of blocking browser access.

tcamise-gpsw, 2024-07-24:
I actually have a CORS proxy Docker image to workaround this. I'll add it as a tool in this repo when I get a chance.

tcamise-gpsw, 2024-07-24:
Well I've added the [CORS proxy tool](https://github.com/gopro/OpenGoPro/tree/main/tools/cors_proxy). Support on the camera is still the ideal solution though so I'll leave this open.

maledies, 2024-07-24:
> It's been specifically blocked for security reasons but perhaps these are outdated / not worth the tradeoffs of blocking browser access.

Honestly I don't see any security gain by not implementing CORS. A malicious site can just forward or link to `http://10.5.5.9:8080/gp/gpControl/command/storage/delete/all`... no SOP applied, just a completely unprotected endpoint executed by a simple browser navigation...

Actual security comes with the authentication required when accessing via HTTPS. And since the `Authorization` header must be actively set per request and is not passively sent by the browser like a cookie, XSRF should be safe and there is no reason to block cross-site requests. Or am I missing something?