gopro/OpenGoPro #576: Switching between preset groups results in unexpected response

George-Gi, 2024-07-27:
When the GoPro is in Video preset group and I send a command to load the Photo preset group, it does work, but at the same time I get back responses for the resolution, fps and lens for the Video preset group. The results seem to default every time to:
- resolution: 4k
- fps: 240
- lens: wide.
Similarly, when the GoPro is in Photo preset group and I send a command to load the Video preset group I get back response for the Photo lens which defaults to:
- lens: wide.

I have registered for updates regarding the video resolution, fps, lens and for the photo lens, but I don't understand when I switch to the other group I get wrong information for the previous group.

tcamise-gpsw, 2024-08-02:
I'm not sure what you mean by "get back responses?" For example, are you querying these settings via BLE / HTTP?

Can you please post some type of log of the communication?

George-Gi, 2024-08-13:
> I'm not sure what you mean by "get back responses?" For example, are you querying these settings via BLE / HTTP?
> 
> Can you please post some type of log of the communication?

I have registered for notification updates over BLE for video resolution, fps, lens and for the photo lens. Whenever I send a command over BLE to change the preset, I get notification updates in the way described in my original question.
The following log is an example when the camera preset is Video and I send a command to change it to Photo.
In the following log we observe that when I send a command to change to Photo preset I get a notification  for video fps and video lens. In this example, the Video preset had an Fps setting different from 240 and Video Lens different than Wide. The Video Resolution was 4k, so I didn't get a notification.
According to my original post and the following log it seems that when I have set any of the video settings (fps, lens, resolution) to a different value other than the apparent default (see defaults in my original post) when I change to photo preset, I receive notifications for the corresponding video settings and the values that I get are the aforementioned "default" ones.

```
Sending msg: [00, 72,0 4, 3E, 02, 03, E9] => Send to message to gopro to switch preset
Msg received: [00, 77, 92, 03, 00] => Receive notification from gopro about fps
Msg received: [00, 77, 92, 7A, 66] => Receive notification from gopro about photo lens
Msg received: [00, 77, 92, 79, 00] => Receive notification from gopro about video lens
Msg received: [00, 77, 93, 60, 00, 00, 03, E9] => Receive notification from gopro about preset group
```

George-Gi, 2024-09-13:
Since this is tagged as Stale now, is there any update?

tcamise-gpsw, 2025-06-06:
This is expected behavior. The Mode Group-specific settings become invalid once the relevant mode group is not active.

closed 2025-06-06 by tcamise-gpsw