gopro/OpenGoPro #577: Star stream often fails Hero 11, response status code 409, camera version 02.30.00

standinga, 2024-07-28:
Hi, I'm integrating GoPro camera support to my iOS app. However I'm often getting 409 status response and when it happens there's no UPD stream starting. In this state when I call stop the response is also 409.

**To Reproduce**
Steps to reproduce the behavior:
Make the HTTP request to `http://10.5.5.9:8080/gopro/camera/stream/start`

**Expected behavior**
The response should be 200 and the UDP stream should start on port 8554

**Screenshots**
```
Optional(<NSHTTPURLResponse: 0x282bb8c00> { URL: http://10.5.5.9:8080/gopro/camera/stream/start } { Status Code: 409, Headers {
    "Content-Length" =     (
        3
    );
    "Content-Type" =     (
        "application/json"
    );
} })
```

**Hardware**
- Camera: Hero11 Black, etc.
- Smartphone if applicable  iPhone6s

**Additional context**
It sometimes works well, it's not clear what would be the exact condition that triggers the error.
Can somebody at least let me know what the 409 means in the context of the GoPro and what could lead to it?

Thanks!!!

standinga, 2024-07-30:
I think the problem was due to the camera stream wasn't always correctly stopped. However, it's not clear why `http://10.5.5.9:8080/gopro/camera/stream/stop` in that state would also result in the same 409 status code.

tcamise-gpsw, 2024-08-12:
Can you please post a log of all of the commands that you are sending to cause this error?

standinga, 2024-08-18:
The command that would trigger the issue was `http://10.5.5.9:8080/gopro/camera/stream/start`. 
It happens if the `http://10.5.5.9:8080/gopro/camera/stream/stop` was not called before disconnecting from the GoPro.

standinga, 2024-08-19:
I closed the issue, the fix is to call the `http://10.5.5.9:8080/gopro/camera/stream/stop` before disconnecting from the GoPro.

closed 2024-08-19 by standinga

TinfoilSubmarine, 2025-05-23:
I am experiencing this issue. When I connect the GoPro to an access point, provision COHN, and try to call `/gopro/camera/stream/start`, it responds with 409. 

I've tried calling `/gopro/camera/stream/stop` before calling `/gopro/camera/stream/start`, but this doesn't fix the issue, I still get 409s when calling start.