gopro/OpenGoPro #578: WirelessGoPro asks for sudo password

grighi96, 2024-07-30:
Hello, I'm implementing a Python script on a Raspberry Pi 3 to control a gopro via wifi.
I tried the gopro-wifi demo both as sudo and standard user and in both case it works. But in the standard user case it asks for sudo password and it accepts even a wrong password.
Then I created my personal Python script and if I launch it as sudo it keeps asking for the sudo password and I get the "Invalid password" error even if the password is correct.
Am I doing something wrong?

grighi96, 2024-07-30:
UPDATE: By enabling the logging I don't get the invalid password error but it still ask the password when launched as sudo.
It still doesn't connect to the gopro via wifi but this might be another type of problem

grighi96, 2024-07-30:
UPDATE: I looked at the log files and in the gopro-wifi demo I can see that when executed as sudo it gets:
        org.freedesktop.NetworkManager.enable-disable-wifi yes
        org.freedesktop.NetworkManager.wifi.scan yes
so it doesn't ask for a sudo password.

When executed as standard user it gets:
  org.freedesktop.NetworkManager.enable-disable-wifi    no
  org.freedesktop.NetworkManager.wifi.scan  auth
and it asks for a password and even if I input a wrong password it works.
On my script I get no and auth both with and without sudo and it doesn't accepts any password.

grighi96, 2024-07-31:
SOLVED: I had to change the NetworkManager policy to allow scan and enable-disable-wifi

tcamise-gpsw, 2024-08-02:
Glad you got it working.
FWIW the way that this is designed to work in demos is that you use the `--password` argument to read your password from stdin.

Anyway I don't see any changes here so will close this.

closed 2024-08-02 by tcamise-gpsw