gopro/OpenGoPro #580: Feature Request: Camera Connect / Wake Using Power USB

sakthivelj, 2024-08-07:
This feature allows the GoPro camera to automatically power on or wake when connected to a USB power source. It enhances the camera's usability by eliminating the need for manual power-on, especially when the camera is mounted in hard-to-reach places.

**Details**
This feature is already available in the GoPro Labs SDK, and implementing it in the OpenGoPro SDK would significantly improve the performance and user experience of GoPro cameras. Integrating this feature ensures consistent functionality across different SDKs provided by GoPro.

**Question**
Can I use the GoPro Labs firmware in conjunction with the OpenGoPro API to achieve this functionality? Please guide compatibility and any potential limitations.

mmarich, 2024-09-09:
I'd like to add my +1 for this feature request. I am developing a USB control solution for an automotive domain, and having the ability to wake the camera without human intervention, but not necessarily start recording, would be a valuable feature.

mmarich, 2024-10-30:
Just an additional though, the optimal use case for my application would be to update the wake setting via HTTP over USB, however I can also see HTTP over WiFi and also being useful. So if I could request a priority in adding this to a control interface, it would definitely be HTTP over USB.

sakthivelj, 2025-01-15:
@tcamise-gpsw Any update available on this?

sakthivelj, 2025-02-28:
Is there any new firmware available for the GoPro Hero 12 to support away wake or power wakeup?

backstromjoel, 2025-03-05:
This would be awesome! I'm trying to implement something where I can wake the cameras after the 8 hour window of bluetooth advertising.

Is there any way of doing this over USB/WIFI?

TimelapseRobot, 2025-06-30:
Need this feature so much for Gopro 11, 12 and 13!