gopro/OpenGoPro #583: Enabling Python GoPro SDK to gracefully handle connection and disconnection based on user's choice after running a script

TarushShankar, 2024-08-13:
I am using GoPro HERO 10 Black with wireless (Bluetooth + Wifi connection) and the appropriate modified demo python scripts for my professional objectives.

Problem Statement:
I am utilising a modified demo video.py file for my objectives. Everytime I run the script, I am realising that my system has to connect with the GoPro on executing a script and it automatically disconnects after running the script. This latency can be problematic since I will have to execute the script consecutively for short video recordings for specific purposes.

Probable Solution to request:
I would want my Wireless GoPro interface to only disconnect from the camera, based on a user entered choice of whether to disconnect or not each time after the script is run. Also, most of the demo python scripts are encoded with the assumption that the script would automatically disconnect the camera after running and has to connect with camera, each time we invoke the python script to run. If we could check for a connection with an interface before automatically connecting the GoPro camera upon invoking the script, that would be helpful too.

Potential Alternatives:
For my professional objectives and repository, I tried to incorporate my specific feature in the WirelessGoPro.py script for the SDK in my local development environment. However, this would be a very individualised modification and it can become redundant upon updating the SDK version for my developing environment.

tcamise-gpsw, 2024-08-30:
Hello. I consider this to fall under the [pinned ticket](https://github.com/gopro/OpenGoPro/issues/487) about connection robustness. Ultimately I'm just not sure that bleak provides the required functionality for this. There is some promising comments in the bleak ticket I linked on that pinned ticket. But really this needs a major investigation.

tcamise-gpsw, 2024-08-30:
I'll leave this ticket open for now but will probably merge it with #487 once (if) this investigation starts.

tcamise-gpsw, 2025-03-25:
The story here is that a `GoProWireless` context session is directly tied to the underlying bleak BLE context session.
That is, closing the `GoProWireless` object will also close the bleak object and thus the BLE connection.

It is very important to close the BLE connection since bleak has no consistent cross-platform solution of establishing communication to a BLE device that is already connected from the OS bluetooth perspective.

I need to do some more searching in the bleak repo to find relevant tickets to link here but I'm fairly certain this request is not currently possible.