gopro/OpenGoPro #584: Updating Hero12 resolution settings over USB yields HTTP 403.

mmarich, 2024-08-18:
**Component**
HTTP API (USB)

**Describe the bug**
Attempting to set the video resolution (the same issue occurs when attempting to set the video frame rate) results in a HTTP 403 error. The same error occurs with or without the command to enable wired control.

**To Reproduce**
Connect to camera via USB, and execute the following CURL commands as per the OpenGoPro documentation:

```shell
user@device % curl -v --request GET --url 'http://172.25.110.51:8080/gopro/camera/control/wired_usb?p=1'
Note: Unnecessary use of -X or --request, GET is already inferred.
*   Trying 172.25.110.51:8080...
* Connected to 172.25.110.51 (172.25.110.51) port 8080
> GET /gopro/camera/control/wired_usb?p=1 HTTP/1.1
> Host: 172.25.110.51:8080
> User-Agent: curl/8.4.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: application/json
< Content-Length: 3
< 
{}
* Connection #0 to host 172.25.110.51 left intact

user@device % curl -v --request GET --url 'http://172.25.110.51:8080/gopro/camera/setting?setting=2&option=9'
Note: Unnecessary use of -X or --request, GET is already inferred.
*   Trying 172.25.110.51:8080...
* Connected to 172.25.110.51 (172.25.110.51) port 8080
> GET /gopro/camera/setting?setting=2&option=9 HTTP/1.1
> Host: 172.25.110.51:8080
> User-Agent: curl/8.4.0
> Accept: */*
> 
< HTTP/1.1 403 Forbidden
< Content-Type: application/json
< Content-Length: 77
< 
{
    "error": 1,
    "option_id": 9,
    "setting_id": 2,
    "available_options": []
}
* Connection #0 to host 172.25.110.51 left intact
``` 

**Expected behavior**
The video resolution is updated. In this case, the requested resolution is for 1080p, although other resolutions supported by the Hero12 also fail.

**Hardware**
- Camera: Hero12
- Version: 2.20

tcamise-gpsw, 2024-08-30:
Hello. My guess is that you are not in a mode that uses the Video Resolution. Can you ensure to enter Video mode by either swiping on the UI or [loading the video preset group](https://gopro.github.io/OpenGoPro/http#tag/Presets/operation/OGP_PRESET_SET_GROUP).

Using the Python SDK, I just tested every resolution listed in the spec and see expected behavior. Here are some snippets reproduced with `curl`:

First a valid resolution option...
```
curl 'http://172.27.102.51:8080//gopro/camera/setting?setting=2&option=100'
{}
```

Now an invalid resolution option...
```
> curl 'http://172.27.102.51:8080/gopro/camera/setting?setting=2&option=102'
{
        "error": 4,
        "option_id": 102,
        "setting_id": 2,
        "supported_options": [
                {"display_name": "5.3K", "id": 100},
                {"display_name": "5.3K", "id": 107},
                {"display_name": "4K", "id": 1},
                {"display_name": "4K 4:3", "id": 18},
                {"display_name": "4K", "id": 108},
                {"display_name": "2.7K", "id": 4},
                {"display_name": "2.7K", "id": 111},
                {"display_name": "1080", "id": 9},
                {"display_name": "4K", "id": 109},
                {"display_name": "1080", "id": 110}
        ]
}
```

mmarich, 2024-09-08:
I can confirm that the camera is indeed in video mode, and I am loading the video preset group prior to attempting to alter video resolution and frame rate.

I do have an older SD card inserted, which causes the camera to display a warning about not supporting higher resolutions / frame rates. Could this be an issue? The camera boots into the 'standard' 4k quality setting, and I am attempting to lower it to 1080p.

mmarich, 2024-09-10:
I can confirm that the second camera exhibits the exact same behaviour:

user@device ~ % curl -v -XGET --url 'http://172.29.146.51:8080/gopro/camera/control/wired_usb?p=1'
*   Trying 172.29.146.51:8080...
* Connected to 172.29.146.51 (172.29.146.51) port 8080
> GET /gopro/camera/control/wired_usb?p=1 HTTP/1.1
> Host: 172.29.146.51:8080
> User-Agent: curl/8.4.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: application/json
< Content-Length: 3
< 
{}
* Connection #0 to host 172.29.146.51 left intact
user@device ~ % curl -v -XGET --url 'http://172.29.146.51:8080/gopro/camera/presets/set_group?id=1000'
*   Trying 172.29.146.51:8080...
* Connected to 172.29.146.51 (172.29.146.51) port 8080
> GET /gopro/camera/presets/set_group?id=1000 HTTP/1.1
> Host: 172.29.146.51:8080
> User-Agent: curl/8.4.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Type: application/json
< Content-Length: 3
< 
{}
* Connection #0 to host 172.29.146.51 left intact
user@device ~ % curl -v -XGET --url 'http://172.29.146.51:8080/gopro/camera/setting?setting=2&option=9'
*   Trying 172.29.146.51:8080...
* Connected to 172.29.146.51 (172.29.146.51) port 8080
> GET /gopro/camera/setting?setting=2&option=9 HTTP/1.1
> Host: 172.29.146.51:8080
> User-Agent: curl/8.4.0
> Accept: */*
> 
< HTTP/1.1 403 Forbidden
< Content-Type: application/json
< Content-Length: 77
< 
{
	"error": 1,
	"option_id": 9,
	"setting_id": 2,
	"available_options": []
}
* Connection #0 to host 172.29.146.51 left intact

pb3975, 2024-09-16:
Getting the same 403 Forbidden results here with Hero 12 black when trying to set either FPS or Resolution options. @mmarich did you identify a solution?

pb3975, 2024-09-16:
Update. I was able to solve this issue by changing controls from "Easy" to "Pro".

mmarich, 2024-09-23:
@pb3975 thanks for the tip. It looks like setting #175 should be set to 1 (Pro) on Hero 11 and newer cameras. I'll give it a try and report back.

mmarich, 2024-09-23:
I can confirm that setting the controls to pro using the HTTP interface does allow the resolution and frame rate settings to be updated. @tcamise-gpsw I would ask that the firmware is updated to indicate this as part of the HTTP 403 response. Does this seem like a reasonable request?

I also did encounter an HTTP 500 error when attempting to set the frame rate on a Hero12 to 60fps after setting the resolution to 1080p. Instead, setting the frame rate to 30fps seems to succeed. I'll investigate this further and possibly create a new ticket if necessary. For now my only request is to provide a meaningful error message.

tcamise-gpsw, 2025-03-25:
This is a major flaw in the current OGP API. That is, there is no way to communicate the dependency hierarchy of settings / options on each other. For one, this hierarchy can be extremely complicated and depend on many settings, etc. It could involve, for example, a ten step process of setting specific settings to specific options.

I don't think this is feasible to return as part of the error message. We need to come up with a solution but it would likely either be:
- auto-set any setting option requested from the client (unlikely)
- provide some (i.e. JSON) file that details the dependencies (likely).

In any case, the easy mode setting here is a major high level dependency for many settings. I'll at least update the docs to note this.

I'll also spawn off a pinned issue to describe this setting dependency issue.