gopro/OpenGoPro #587: Wrong IP address

kobezf, 2024-08-29:
I have a GoPRO camera HERO11 Black. I am trying to start live stream in a restricted network with the limited access to the internet. I noticed, that camera sends request to api.gopro.com before starting stream to check the internet connection. So, I added it to my DNS server, but camera uses another ip address instead of it. Here is my tcpdump:

> 10	Aug 27, 2024 12:04:48.112902000 CEST	7.385854	10.10.10.59	10.10.10.1	DNS	73	Standard query 0x734e A api.gopro.com
11	Aug 27, 2024 12:04:48.113806000 CEST	7.386758	10.10.10.1	10.10.10.59	DNS	89	Standard query response 0x734e A api.gopro.com A 10.0.6.30
12	Aug 27, 2024 12:04:48.125266000 CEST	7.398218	10.10.10.59	10.0.0.0	TCP	74	42396 → 80 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=2602477525 TSecr=0 WS=32

So you can see, that DNS server replies with 10.0.6.30, but the camera uses 10.0.0.0.

Do you know why?

tcamise-gpsw, 2024-09-10:
Hello. There is a new feature coming in the next Hero 13 firmware update that will allow you to connect a network without internet access which should solve your problems here.

I'll leave this open until that feature is released.

kobezf, 2024-09-13:
@tcamise-gpsw are you planning to do this firmware for HERO 11 and 12?

tcamise-gpsw, 2024-09-13:
Currently the answer is no. But if there are future H11 / H12 firmware updates it is likely that this will be included.

kobezf, 2024-10-03:
@tcamise-gpsw I got a HERO 13 and I have a problem: the camera doesn't see my WiFi hotspot that I created on my Linux device. On the other hand, I tried to create a hotspot from my mobile phone and it sees it. Are there any requirements for the WiFi network?

kobezf, 2024-10-08:
Also, I found another issue, camera is able to connect to the WIFI with the internet access, but when I try to do so with restricted network, GOPRO is not able to connect to it. Here is my log:
```
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'request control'
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': sending SetControl command
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing request control response
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': got set control response: 1
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'start scan'
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': sending StartScan request
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing start scan response
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': received StartScan response: result 1, scanning state 2
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'wait start scan'
Oct 08 13:26:55 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': waiting for start scanning
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wait start scan response
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': got response start scanning with scan id 9, total entries 18, cameras's provisioned SSIDs 1
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'get ap entries'
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': sending GetAPEntries command
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing get ap entries response
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': got message result 1, scan id 9, amount entries 16
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': found wifi TestKPNsimcard ssid
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': found TestKPNsimcard ssid with prov state false
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'connect wifi'
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': sending connect new wifi command
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing connect wifi response
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': got message result 1, provisioning state 2, timeout 20 seconds
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wifi request 'wait prov. state'
Oct 08 13:27:00 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': waiting for provisioning state response
Oct 08 13:27:02 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': processing wait prov. state response
Oct 08 13:27:02 live-stream[2261]: [debug] 'EF:33:39:DF:51:CF': received provisioning state response 7
Oct 08 13:27:02 live-stream[2261]: [warning] 'EF:33:39:DF:51:CF': cannot connect to wifi 7, reconnecting...
```
I also tried to do so through the QuickAPP, but still, cameras wasn't able to connect to the restricted network

kobezf, 2024-10-25:
@tcamise-gpsw Do you have any idea, why it does not work?

kobezf, 2025-01-24:
@tcamise-gpsw  finally it was fixed and it works.