gopro/OpenGoPro #593: Update OpenGoPro camera list to include serial number prefixes

mmarich, 2024-09-09:
It would be useful to also list the serial number prefixes for each supported camera model listed here:

[https://gopro.github.io/OpenGoPro/](https://gopro.github.io/OpenGoPro/)

Serial number prefixes are currently provided in a GoPro community article here:

[https://community.gopro.com/s/article/how-to-identify-your-gopro-camera?language=en_US](https://community.gopro.com/s/article/how-to-identify-your-gopro-camera?language=en_US)

The community article does not list values for the Hero13. Maintaining this information as part of the OpenGoPro documentation would be valuable to developers.

tcamise-gpsw, 2024-09-10:
This is a good idea. Although I will add it in the individual BLE / HTTP tables so as not to clutter the home page.

tcamise-gpsw, 2025-03-25:
The serial prefixes are not available in the [BLE supported cameras table](https://gopro.github.io/OpenGoPro/ble/#id1)

closed 2025-03-25 by tcamise-gpsw