gopro/OpenGoPro #597: Hero 13 cannot pair with Windows 11

demiantres, 2024-09-11:
The Hero 13 has the same problem as the Hero 12 had (https://github.com/gopro/OpenGoPro/issues/476, https://github.com/gopro/OpenGoPro/issues/395). I am writing "had" because one of the Hero 12 firmware updates fixed the problem.

In short: The Hero 13 cannot pair with Windows 11. The workarounds which allowed to pair the Hero 12 prior to the firmware update, do not work with the Hero 13. Pairing (without the Quik app) works fine on Android, iOS, and macOS.

tcamise-gpsw, 2024-09-11:
Can you clarify if my assumptions about this are correct?
1. The problem is actually _re-connecting_; not _initial pairing_. That is, subsequent connections after the initial pair can not be established without selecting "Pair Device" in the camera UI
2. This is only an issue for you on Hero 13; not Hero 12.
    - If this is true, then I need to close #476. I'll do some testing of this when I get a chance

demiantres, 2024-09-11:
1.) Actually, with the Hero 13, it is a problem with pairing (I cannot pair the camera on Windows at all).
I also did some more testing, and on macOS it seems as if the camera has to be in the pairing state to get it connected again.

demiantres, 2024-09-11:
I tested everything again:

Hero 12 (firmware 2.20):
- Connection on Windows 11 works fine.
- Connection on macOS works fine.

Hero 13:
- Pairing does not work on Windows 11.
- Connection on macOS works only when the camera is in "Pairing" mode.
- Connection on iOS/Android works fine.

tcamise-gpsw, 2024-09-12:
I just tested this on Hero 13 using the `gopro-photo` demo from the Python SDK 0.17.0, factory resetting before testing each OS. Results are as follows:

| OS | Initial Pairing | Immediate Reconnection after clearing pairing | Reconnection from sleep |
|-|-|-|-|
|Windows 10|✅|✅|✅|
|Windows 11|✅|✅|✅|
|MacOS|✅|❌  discovered via scanning but connection attempts fail to establish. connection works immediately after selecting “Pair Device” on U|❌discovered via scanning but connection attempts fail to establish|
|Linux (RPi)|✅|✅|✅|

This is pretty drastically different than what you are seeing which I can not explain. In any case it appears to exhibit the same behavior as Hero 12

tcamise-gpsw, 2024-09-13:
I'm going to update the ticket name to reflect my test results here.

KonradIT, 2024-09-13:
Can confirm a HERO13 Black cannot pair to Windows 11, at least when it has been paired to an Android phone / RPI before.

Windows OS: Version	10.0.22631 Build 22631

HERO13: v01.10.00 

Windows add device dialog does not see gopro at all, even though the Hero13 can see the windows device after clicking Pair Device.

Clicking on the name of the windows machine on the gopro makes an attempt to pair but doesn't succeed.

tcamise-gpsw, 2024-09-13:
Oh I'm realizing now that you all are just using the System menu. That is not expected to work. 

What is the use case here? What are you going to do after getting it connected in this manner?

KonradIT, 2024-09-13:
This used to work on earlier cameras, even Hero10 iirc. Is it just not needed and the BLE python code will handle the connection?

demiantres, 2024-09-14:
Actually, I tried both the Windows Bluetooth manager (just out of curiosity),  and my own connection code. My own code sees the camera, but once I start the connection process the GoPro' pairing screen displays a Bluetooth symbol followed by three dots (with other systems "Quik" is shown instead of the dots) and then the connection process stops.

demiantres, 2024-09-14:
Ok, final tests: Hero 12 & 13 work both without problems on Windows 11.
On macOS Hero 13 works only in pairing mode (as described in your post above). Hero 12 works fine one macOS.

mohammad-askari, 2024-09-30:
Same problem here on macOS. Are there any updates?

MattL-NZ, 2025-02-11:
> I just tested this on Hero 13 using the `gopro-photo` demo from the Python SDK 0.17.0, factory resetting before testing each OS. Results are as follows:
>
> OS	Initial Pairing	Immediate Reconnection [...]

Is there any update for MacOS @tcamise-gpsw? I am still seeing results where the reconnect won't work without first selecting "pairing mode" on my GoPro Hero 13

hfznr, 2025-03-05:
The 12 Black works well when I pair the device with macOS. I can control the camera even when it is turned off after pairing using the SDK. However, the same code does not work with the 13 Black. Has anyone solved this issue? Do you have any suggestions?

You can try the following steps to get the same error:  

- Connect the devices from your software and pair them, then pair the devices with your phone. 
- Afterward, turn off the devices and turn them on again from your phone.
- Do not go to 'pair device' again. 
- Instead, directly run the software on macOS and try to start recording. 

The 12 Black starts recording, but for the 13 Black, I need to select 'pair device' again to start recording.

MattL-NZ, 2025-03-05:
I am experience the exact same results as @hfznr. Would be great to have a resolution to this.