gopro/OpenGoPro #600: Question: STA mode and password change

demiantres, 2024-09-13:
What is the intended way of handling Wi-Fi network password changes (the router's SSID stays the same, but the password has been changed) in STA mode (https://gopro.github.io/OpenGoPro/ble/features/access_points.html)?

One way could be to use ``RequestConnectNew`` everytime, but the docs clearly state that this should function should only be used with non-provisioned networks.

tcamise-gpsw, 2024-09-13:
You do need to use `RequestConnectNew` in this case. I'll need to update the docs to note this special case.

demiantres, 2024-09-14:
As a side note: I have completely abandoned using ``RequestConnect`` in favor of ``RequestConnectNew`` because ``RequestConnect`` is quiet unreliable.