gopro/OpenGoPro #603: Hero 13 webcam mode compatibility

hugolegendre, 2024-09-16:
**Component**
 Specs (BLE, HTTP)

**Describe the bug**
I have a python program that allows my GoPro Hero 12 to stream as a webcam via COHN on my private network. I use https://{ip_address}/gopro/webcam/start as an URL to start the webcam mode and it works perfectly. But I just received the new Hero 13 Black and I got HTTP 500 internal error when I tried to use the same webcam mode. Is there a compatibility issue with the new GoPro?

**To Reproduce**
Steps to reproduce the behavior:
1. Test Webcam HTTP requests on Hero 13 via COHN

**Expected behavior**
Webcam Mode via COHN that works with GoPro HERO 13

**Hardware**
- Camera: Hero13
- Windows & Python

bfreeland-gpsw, 2024-10-01:
Hi [hugolegendre](https://github.com/hugolegendre),

I'm not able to replicate this problem on the most recent firmware. How persistent is the issue for you? Was it a one time problem that went away after rebooting the camera or are you seeing it consistently?

tcamise-gpsw, 2025-01-07:
This was a [known issue](https://literate-couscous-wl75qnk.pages.github.io/ble/features/settings.html#setting-ids) for Hero 13 initial firmware but it should be fixed now. 

@hugolegendre are you still seeing this problem?

closed 2025-02-17 by github-actions[bot]