gopro/OpenGoPro #609: RTMP Streaming Programatically

martincarapia, 2024-09-23:
So, I'm trying to do something similar where I set cameras to an RTMP server to be used for streaming programmatically with the python SDK. I've got to the point where I can configure the cameras to an access point and configure it to live stream, but I can't start live streaming. The docs [here](https://gopro.github.io/OpenGoPro/ble/features/live_streaming.html) point that you need a GoPro 12 or higher to ping for Livestream Ready Status. Why can't Gopros >9 respond with Ready Status?

 I'm confused that you can set up live streaming from your phone using the GoPro Quik but not from the SDK. 

I find it even more weird that there was a livestreaming demo before (PR #318) the GoPro Hero 12 release of September 15, 2023. 

Why did adding Hero 12 support kill live streaming for other models? 

Essentially my question is, do I need to buy a GoPro 12 in order to complete my project or is there a work around for older models.

_Originally posted by @martincarapia in https://github.com/gopro/OpenGoPro/discussions/428#discussioncomment-10729204_

bfreeland-gpsw, 2024-10-01:
Hi [martincarapia](https://github.com/martincarapia),

Hero 12 introduced support for querying livestream status over HTTP. You should be able to use the BLE endpoint for querying livestream status on Hero 9. It looks like when Hero 12 was released it looks like older camera models were removed by accident from the BLE livestream endpoint documentation. We'll try to get that fixed in the next documentation update.

martincarapia, 2024-10-01:
Thank you so much, I digged through the docs and tested myself and figured it wasn't really an issue. I ended up starting to develop a solution to my problem. Fixing the documents should definitely be a priority though. 
https://github.com/martincarapia/GoProStreamRelay

closed 2024-10-01 by martincarapia

NathanRignall, 2025-02-16:
Hello, has the documentation been updated to include the http endpoints to start/stop/configure rtmp streams?

I am looking to use GoPros for streaming and would like to be able to start and stop remotely but do not have access to BLE.